OrcaSlicer/OrcaSlicer #1306: Printer Add: Creality non-S series

Azuroth2013, 2023-06-09:
Would like to see the non S series of creality Ender printers. Such as the Ender 3 Pro.

karcaw, 2023-11-06:
What does it take to add a new Profile?  I may be able to work on it.

karcaw, 2024-02-12:
but its still an issue...

cochcoder, 2024-03-12:
I am currently making some profiles for the Ender 3 Pro, would you say that the same settings used for the non-pro Ender 3 work good for the pro version? I can't find any differences online and I only have a Ender 3.

karcaw, 2024-03-12:
yes, i based my profile on the ender 3 s1 bits that were in there, and made sure the motion ability numbers matched what I could see in Cura and Prusa....

cochcoder, 2024-03-12:
I'll make a PR, for the Ender-3 Pro, as soon as the one I have right now gets merged

cochcoder, 2024-03-19:
I have opened a PR that copies the in-built profile for the Ender-3 and uses it for the Ender-3 Pro. I did this as I can't find any information online that gives settings significantly different than the current Ender-3 profile.

closed 2024-06-26 by github-actions[bot]