OrcaSlicer/OrcaSlicer #1308: Absolut Extrusion create to much volumetric flowrate

abstractFlo, 2023-06-09:
**Describe the bug**
If i enable absolut extrusion inside the printer settings and slice my model, the volumetric flow rate is unexpected high. I have ignore this for some time, but my print ist canceld about 58%. Klipper says to much cross section extrusion. I have default a value of 15, but klipper wants 83 before canceled. If i activate relative extrusion, all is fine.

The preview of the sliced model shows the right volumetric flow rate.

**3mf File for This Bug**
[wrong_volumetric_flowrate.zip](https://github.com/SoftFever/OrcaSlicer/files/11709288/wrong_volumetric_flowrate.zip)

**To Reproduce**
uncheck relative extrusion inside the printer settings

**Expected behavior**
Same volumetric flow rate no matter which extrusion mode is set.

**Screenshots**
[image]
[image]

**Printer model**
Voron 0.2

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Orca Version: 1.6.3

abstractFlo, 2023-06-09:
It's only for this model in my case, any other model works perfectly with absolut extrusion mode

Habag111, 2023-06-22:
[image]
[image]

I'm experiencing the same issue. It's so bad klipper errors out "Stepper too far in past". Using absolute extrusion eliminated the problem. I'm using orca 1.6.3 on windows 10. The .stl I used was kevinakasams tolerance test https://github.com/kevinakasam/KlackEnder-Probe/blob/main/STL/Tolerance_Test.stl

justr0st, 2023-06-23:
> Using absolute extrusion

You mean using **relative** extrusion

closed 2023-11-22 by github-actions[bot]