OrcaSlicer/OrcaSlicer #1309: Set process preset per-plate

shayded-exe, 2023-06-09:
**Is your feature request related to a problem? Please describe.**
I like to set up multiple build plates for different parts in the same project. Usually I'm putting multiple of the same part on a plate and I want each plate to print with its own settings.

Currently I either have to remember to switch presets when changing plates, or set per-object settings on every object on the plate which is extremely arduous and not easily updated because you cannot set per-object presets (only individual settings).

**Describe the solution you'd like**
In the plate settings dialog where you can name the plate, I would like to be able to select a process preset.

julie777, 2023-06-09:
I have another use case. Often while iterating on a design, I export a part from Fusion 360 and then import it into Orca. Because the project has many plates, I have the set the specific to the new object to the object. 

When I am ready to iterate again, I delete the object from the plate and then import the new STL. Because settings were attached to the object, I now have to got through the process to set all the settings again, for the new instance of the same object.

shayded-exe, 2023-06-09:
> I have another use case. Often while iterating on a design, I export a part from Fusion 360 and then import it into Orca. Because the project has many plates, I have the set the specific to the new [...]

You can actually get around this by right clicking on your object in the slicer and doing "Replace with STL"! Or if you just overwrite the same file you can do "Reload from disk".

hliebscher, 2023-06-10:
@shayded-exe  i use multiple instances with different print settings so it is also possible use different printers at the same time and easy name of the 3mf like parts or settings...

julie777, 2023-06-19:
@shayded-exe Thanks!

closed 2023-11-22 by github-actions[bot]