OrcaSlicer/OrcaSlicer #1314: height range modifier

StephInoP, 2023-06-10:
**Describe the bug**
I added a height range modifier to the model, set it up and done. When re-open the slicer, height range modifier is gone.
Not in the project anymore. I have tried it many times, so i confirm there is a problem here.

**Printer model**
Bambulab X1 Carbon

**Desktop (please complete the following information):**
Mac OS Monterey

Stargo44, 2023-06-10:
Same issue with me height range modifie are not saved or not loaded.
 Windows 10

foto-boss, 2023-06-10:
I confirm. Ranges are not saved.

hliebscher, 2023-06-10:
same as #1195  #1071

OmiCron07, 2023-06-27:
Bambu Lab is able to save and load the height modifier. Orca does not save the height modifier, nor load it even if present.

Bambu Lab:
[image]

Orca:
[image]

MotoFreak, 2023-07-18:
having the same issue as well just like the others.  Just add a height modifier and save and reload project, and its gone!

closed 2023-11-22 by github-actions[bot]