OrcaSlicer/OrcaSlicer #1320: Cannot Turn Off Timelapse Setting.

PBCups, 2023-06-10:
**Describe the bug**
Under the timelapse setting there is no "none" option. Only Smooth and Traditional.

**To Reproduce**
I have tried turning on advanced settings and nothing changes, it doesn't matter what file I am slicing. I have restarted my computer as well and still no option for "none".

**Expected behavior**
The ability to turn off timelapses from the slicer.

**Screenshots**
https://imgur.com/a/1Wd9wgf

**Printer model**
Sovol SV06
**Desktop:**
 - OS: Windows
 - Version: 11

RaySajuuk, 2023-06-11:
This option shows the two methods of how timelapse video is generated. The difference is that the smooth method moves the toolhead to the purge chute every time before snapshots.

Timelapse on and off option is in the send print job window.
[image]

I see you are not using the Bambu printer, I don't know if you have these options.

PBCups, 2023-06-11:
I understand what both methods do, however there is supposed to be a 'none' option in addition to those two options. I do not 

> This option shows the two methods of how timelapse video is generated. The difference is that the smooth method moves the toolhead to the purge chute every time before snapshots.
>
> Timelapse on and [...]

I know what both time lapse options do. There should be a third option, 'none'. Everything I've looked up says there should be three options and I only have two. 

As far as the send print menu there is only 'print' (which doesn't work for my printer) and 'export g-code' neither of these has anything to do with time lapse.

PBCups, 2023-06-11:
I did not mean to close this issue, I am still having this problem.

RaySajuuk, 2023-06-12:
If my memory serves me correctly, classic and smooth are only two options. I don't see why a "none" option is needed.
This option is only to set the method of time-lapse generation, not enable to disable it.

It is like you are driving a car forward or reverse. You change gears for "Drive" and "Reverse" but still use the acceleration paddle for moving the vehicle. Therefore, changing only the gear without the acceleration paddle is meaningless if your is to drive it.

This is why "none" is not an option in that setting because that setting has no meaning if you don't enable timelapse.

Furthermore, OrcaSlicer is based on BambuStudio, which is specialized in working with Bambu Lab printers, with support for other printer manufacturers.

I assume the timelapse option is not showing for your printer, or the option does not do anything with your printer.

PBCups, 2023-06-12:
You are correct that the time lapse doesn't do anything for my printer. However, I use a beagle cam. The camera has a time lapse feature that I was using for every print until I switched to Orca. Now my time lapse doesn't work. Probably because now my slicer is trying to put its own time lapse in that my camera doesn't recognize. 

I need to turn off time lapse on Orca so I can actually use time lapse from my camera. 

I've seen lots of screenshots of the time lapse drop down in orca with a 'none' setting. I know it's possible, and that's how it should be, yet it's not like that for me.

pietchaki, 2023-07-28:
@PBCups Not sure about your specific configuration, but I'd say you have to insert the timelapse gcode/macro in the "Before Layer Change" section of the printer settings.

Now, as a improvement for OrcaSlicer, I'd say it would be best to automatically disable the Bambu specific options if the current printer is not a BambuLabs printer, and add "BambuLabs Printer exclusive" in the option description.

closed 2023-11-22 by github-actions[bot]

otede, 2024-06-14:
Same issue here. Some poor design there.