OrcaSlicer/OrcaSlicer #1321: line width depending on nozzle width

delStipe, 2023-06-10:
**Additional context**
Is it possible to set the width of the line to depend on the percentage of the width of the nozzle, as solved in SuperSlicer.
  Then those parameters would change automatically...

Hvala :)
  
[image]

jeremytodd1, 2023-06-11:
Fully agree. How SS handles the percentages is much better than how PS handles the percentages.

With SS I can easily make profiles for different nozzle sizes. With PS it's a pain, to the point where I don't even bother.

MaikelKelder, 2023-06-16:
I second this, this is for me the main reason to keep using SS, since in Orca I dont like this fixed setting.

julie777, 2023-06-19:
> Fully agree. How SS handles the percentages is much better than how PS handles the percentages.
> 
> With SS I can easily make profiles for different nozzle sizes. With PS it's a pain, to the point where I don't even bother.

@jeremytodd1 
Just curious, do you mean you can "easily make profiles for different nozzle sizes" or can you use the same profile because line sizes change automatically based on nozzle size?

MaikelKelder, 2023-06-19:
The last, changing the nozzle in printer setting sets automatically the
line widths in print settings.

closed 2023-11-22 by github-actions[bot]

jagerman, 2023-11-22:
Bad bot.