OrcaSlicer/OrcaSlicer #1330: Unnecessary fan speed commands when forcing overhang cooling

quantumfrost, 2023-06-12:
The slicer emits unnecessary back-to-back M106 commands when `Force cooling for overhangs and bridges` is set. This causes the hotend to stutter on some printers.

It seems like the current logic is to set fan speed back to normal at the end of an overhang line. If it's followed immediately by another overhang line, another M106 is emitted. The logic should only emit M106 if a fan speed change is needed for the next line.

[image]

quantumfrost, 2023-06-14:
If anyone else is facing this issue, I've created [a post-processing script](https://gist.github.com/quantumfrost/bf45a516f707212c9e0fd64d2c7e0293) that will remove unnecessary fan speed changes.

Download and add the path at the bottom of the "Others" tab under Process:
`"/path/to/python" "/path/to/cleanup_m106.py"`
This code is provided without any warranties, but it seems to work fine. In one instance it removed over 10k M106s.

Some more info about the bug: it only happens when "Cooling overhang threshold" is set to <=25%, and the stuttering only appears to happen on Marlin, not Klipper.

wolfie, 2023-08-02:
I have this same thing happen, but with Klipper (mostly stock RatRig VCore3 profile). I have not done any further diagnostics on whether or not the cooling overhang threshold leads to this or not.

closed 2023-11-22 by github-actions[bot]

quantumfrost, 2024-01-04:
This is not fixed