OrcaSlicer/OrcaSlicer #1332: Don't clear the slice preview when changing a setting until the new preview is generated

shayded-exe, 2023-06-12:
**Is your feature request related to a problem? Please describe.**
When changing a setting, the sliced model preview is replaced with a placeholder until you re-slice. This is annoying when changing multiple settings if you'd like to continue to look at the previous slice. This isn't a big deal on smaller models that slice in <5s, but it can be quite annoying for bigger models. It makes it more difficult to compare settings as opposed to seeing the preview immediately change from the old to the new.

**Describe the solution you'd like**
Instead of replacing the slice preview with a placeholder after a setting has changed, instead show either an indicator or notification that the preview is stale. Once the new preview is available, immediately swap it with the old one.

**Describe alternatives you've considered**
One alternative is to just take screenshots and compare them. This is likely what most people do, however it's still a more complicated workflow that is left up to the user to figure out.

What I'm really trying to achieve is a convenient and easy way to compare different slicer settings. The feature I proposed doesn't _really_ address this, it merely makes it easier to do small A/B comparisons. Considering that 90% of the time you spend in the slicer is tweaking and comparing settings, we should absolutely have a more convenient way of doing that. I'll post a separate feature request to discuss a more fleshed out way to easily compare slices.

julie777, 2023-06-19:
I think my request is for the same issue. I would like to be able to turn on auto slice (like Prusa slicer), but not in preferences because I do not want re-slicing to happen on every setting change. 

I would like to have a auto slice checkbox somewhere, so when I am trying to compare slicing with different settings I can change a setting and have it re-slice automatically. It already maintains the preview display type and layer being displayed. However, I have to change the setting and then click on "Slice Plate".

shayded-exe, 2023-06-20:
@julie777 My request is a bit different. I want the previous slice preview to remain until the new preview is actually ready. That way you can more easily see what changes.

FinalFrag, 2023-11-04:
I've bumped into this too. Would be a very nice addition.

closed 2024-02-19 by github-actions[bot]