OrcaSlicer/OrcaSlicer #1337: Purge/wipe towers

brueter, 2023-06-13:
### Discussed in https://github.com/SoftFever/OrcaSlicer/discussions/1336

<div type='discussions-op-text'>

<sup>Originally posted by **brueter** June 13, 2023</sup>
Wondering if theres any possibility of adding purge/wipe towers similar to what superslicer has. For MMU units like the ERCF on units without anywhere to purge filament other than the plate having a purge tower seems like a good solution. Superslicer has this implemented already and has automatic volume calculation based on "pigmentation" values for each filament. This pigmentation calculation is nice but is very difficult to calculate compared to how simple calculating purge volume is currently in bambu/orca. If these volumes could be used to automatically calculate the purge tower volumes it would be ideal.

Below is the code I believe superslicer is using for their tower generation, let me know if this isn't correct and I can try and find the correct reference.
https://github.com/supermerill/SuperSlicer/blob/master/src/libslic3r/GCode/WipeTower.cpp
https://github.com/supermerill/SuperSlicer/blob/master/src/libslic3r/GCode/WipeTower.hpp
https://github.com/supermerill/SuperSlicer/blob/master/src/slic3r/GUI/WipeTowerDialog.cpp
https://github.com/supermerill/SuperSlicer/blob/master/src/slic3r/GUI/WipeTowerDialog.hpp</div>

sorted01, 2023-06-18:
I was looking if i can find a request like that. Thank you for bringing this up. I would love that feature <3

Surion79, 2023-06-25:
I would love that feature as well <3

jeanres, 2023-06-27:
I would love this feature as well

adolfotregosa, 2023-07-01:
Count me in too. Wipe tower would be great a addition.

Nofuss73, 2023-07-03:
I would love this feature too..

Cheo7811, 2023-07-04:
I always wondered why it was not present since it is implenented in Prusa slicer and Superslicer

petrude, 2023-07-12:
I would like this feature as well. Orca Slicer is an awesome slicer.  Do not want to switch to another slicer to use ERCF.

brueter, 2023-07-12:
Seems like this gas gotten a lot of activity and would benefit a lot of people, is there a good way to get in contact with devs to possibly get this going?

> ### Discussed in #1336
> Originally posted by **brueter** June 13, 2023 Wondering if theres any possibility of adding purge/wipe towers similar to what superslicer has. For MMU units like the ERCF on units without anywhere to purge filament other than the plate having a purge tower seems like a good solution. Superslicer has this implemented already and has automatic volume calculation based on "pigmentation" values for each filament. This pigmentation calculation is nice but is very difficult to calculate compared to how simple calculating purge volume is currently in bambu/orca. If these volumes could be used to automatically calculate the purge tower volumes it would be ideal.
> 
> Below is the code I believe superslicer is using for their tower generation, let me know if this isn't correct and I can try and find the correct reference. https://github.com/supermerill/SuperSlicer/blob/master/src/libslic3r/GCode/WipeTower.cpp https://github.com/supermerill/SuperSlicer/blob/master/src/libslic3r/GCode/WipeTower.hpp https://github.com/supermerill/SuperSlicer/blob/master/src/slic3r/GUI/WipeTowerDialog.cpp https://github.com/supermerill/SuperSlicer/blob/master/src/slic3r/GUI/WipeTowerDialog.hpp

brueter, 2023-07-12:
Seems like this is getting a lot of attention and would help a lot of people, is there a good way to possibly get in contact with the devs?

jeanres, 2023-07-28:
I am happy to take a whirl at implementing this, if the devs are ok with this being done?

brueter, 2023-07-28:
> I am happy to take a whirl at implementing this, if the devs are ok with this being done?

Don't see why they wouldn't want it implemented, if you need any help with testing feel free to reach out on discord, my username is zchem

jeremytodd1, 2023-08-21:
I'm also looking forward to having a purge block feature brought to Orca.

I plan to build a Tradrack later in the year. Knowing that I wont currently be able to use Orca for it though, because of the lack of a purge block,  is not a good feeling.

brueter, 2023-10-06:
Looks like this may have been added with the "purge in prime tower" option, correct me if I'm wrong but if this is the case I can close this as completed!

Surion79, 2023-10-06:
The functionality requires to calculate correct purge tower size depending on purge settings. If the tower has always the same size for all kind of prints, the functionality is not what is requested. So please do not close this until it is verified by multiple ercf users.Thanks

brueter, 2023-10-06:
Looks like you may be correct, hopefully this will be easier with this new update though!

Surion79, 2024-01-05:
Still requesting this festure. It is important for Single Extruder MMU

CristianGTR97, 2024-01-23:
Please add this feature

brueter, 2024-04-23:
Still wondering if this can be implemented, does anyone have ideas for getting this done?

igiannakas, 2024-05-01:
I believe this already is implemented:

[image]

[image]

The purge tower changes size depending on the purge volume needed. The purge volume needed can either be entered in the filament flushing volumes table or auto calculated based on filament colour.

See above - reducing the purge values reduces the amount of material on the prime tower. Same for the opposite. Clicking recalculate calculates these values based on colour.

igiannakas, 2024-05-01:
There was a bug where the flush volumes were not displayed correctly in the prime tower statistics but this PR here fixes it: https://github.com/SoftFever/OrcaSlicer/pull/5223

closed 2024-08-07 by github-actions[bot]

CombatVolcano, 2025-01-15:
Please. Do anything to improve this. 
Why do you hide the 'purging volumes' button EVER AT ALL EVER? EVER? EVER?

igiannakas, 2025-01-15:
It’s not hidden - it’s in the flushing volume settings. What am I missing?