OrcaSlicer/OrcaSlicer #1339: As of firmware 01.05.02.00 Playing Video will resize video window and push AMS controls off screen

ChaosBlades, 2023-06-13:
**Describe the bug**
As of the firmware version that released today 01.05.02.00 starting the live camera will slightly widen the video player and cut off the AMS window. I tried closing and re-opening OrcaSlicer with the same result.

**To Reproduce**
1. Upgrade to latest firmware
2. Start Live View

**Screenshots**
[image]

**Printer model**
Bambu Labs X1 Carbon with AMS

**Desktop (please complete the following information):**
 - OS: Pop!_OS 22.04 LTS x86_64
 - Version 1.6.3

ChaosBlades, 2023-06-13:
After starting another print I can't reproduce the issue. This issue may only occur on multi color prints. I'll keep testing.

igiannakas, 2023-06-17:
> After starting another print I can't reproduce the issue. This issue may only occur on multi color prints. I'll keep testing.

Its happened to me before on MacOS even with the previous firmware - the firmware version has nothing to do with this but rather the gui layout.

ChaosBlades, 2023-06-19:
I'm not able to reproduce this anymore even on multi color prints.

I'll leave it up to SoftFever on whether this get closed or or left open. It might be a Bambu issue.

closed 2023-11-21 by github-actions[bot]