OrcaSlicer/OrcaSlicer #1344: Add printer screen text not displaying correctly on Ubuntu (Not a problem on Windows 10)

e-dreyer, 2023-06-14:
**Describe the bug**
With the Ubuntu implementation the text in the "Add physical printer" scree is invisible or just not there. I as an experienced user from PursaSlicer and using OrcaSlicer on Windows 10, could still use it. But it could give problems for other users

**To Reproduce**
Add a new physical printer in Linux Ubuntu using the lastes AppImage and the text will be missing.

**Expected behavior**
That the missing label explaining the required input to be visiable

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Printer model**
Creality CR6-SE (I don't think it is related to my printer or Klipper setup)

**Desktop (please complete the following information):**
 - OS: Linux
 - -  Version: Ubuntu 22.04.2 LTS

neclimdul, 2023-11-13:
Tested this in 1.8.0 rc2 and its still broken. Can't tell what I'm suppose to add to the form to make it work.

StayPirate, 2024-01-27:
This is happening to me as well on Archlinux and the official Appimage

iromero91, 2024-04-15:
still happens on Debian 12 and the latest official Appimage, Forcing light mode just makes different text be invisible.

StayPirate, 2024-06-23:
v2.1.0 fixed this problem for me on Archlinux.  If other people here confirm, then this issue can be closed.

neclimdul, 2024-06-24:
Waiting for the appimage fix or the flatpak release then I'll test it. Kinda of hard to test on Ubuntu like the original report ATM.

closed 2024-09-30 by github-actions[bot]