OrcaSlicer/OrcaSlicer #1346: Crashing when slicing Lithophane

Festivejelly, 2023-06-14:
**Describe the bug**
When I try to slice something complex like a lithophane OrcaSlicer just crashes without warning.

**3mf File for This Bug**
I cant provide this as I dont own the rights to the file

**To Reproduce**
Stack a few lithophanes vertically and try to slice.

**Expected behavior**
Slicing

**Screenshots**

**Printer model**
X1C

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 10
 
 Contents of crash log file:
 
 Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0xffffffff 
Context :0000005337AF2E40 
ContextFlag : 0x10005f, EFlags: 0x10202 
Fault address:  0x676E131F 0x0:0x0 

Show CallStack:

Call SymGetLineFromAddr64 ,Address 676e131f Error:000001e7

Call SymGetLineFromAddr64 ,Address 67aaaed1 Error:000001e7

Call SymGetLineFromAddr64 ,Address 67aaaa37 Error:000001e7

Call SymGetLineFromAddr64 ,Address 67b93fa6 Error:000001e7

Call SymGetLineFromAddr64 ,Address 67b88b53 Error:000001e7

Call SymGetLineFromAddr64 ,Address 67a9d81c Error:000001e7

Call SymGetLineFromAddr64 ,Address 678a998c Error:000001e7

Call SymGetLineFromAddr64 ,Address 6789e6ec Error:000001e7

Call SymGetLineFromAddr64 ,Address 678b712b Error:000001e7

Call SymGetLineFromAddr64 ,Address 6877abfe Error:000001e7

Call SymGetLineFromAddr64 ,Address 6877416d Error:000001e7

Call SymGetLineFromAddr64 ,Address 6877d592 Error:000001e7

Call SymGetLineFromAddr64 ,Address 6877ec59 Error:000001e7

Call SymGetLineFromAddr64 ,Address 6877ee9d Error:000001e7

Call SymGetLineFromAddr64 ,Address 0c221bb2 Error:000001e7

Call SymGetLineFromAddr64 ,Address 0c8e7614 Error:000001e7

Call SymGetLineFromAddr64 ,Address 0e5026a1 Error:000001e7

Festivejelly, 2023-06-14:
This only seems to be happening if I select to print with a raft

smolz, 2023-07-01:
Same, try to print a raft it crashes.  As a note it also crashes on the latest version of BambuStudio

closed 2023-11-21 by github-actions[bot]