OrcaSlicer/OrcaSlicer #1349: Orca Slicer - Newer 3mf version

ETS5, 2023-06-14:
**Describe the bug**
The 3mf's version 1.6.2.4 is newer than Orca Slicer's version 1.6.0.66, Found following keys unrecognized:
-extruder_clearance_max_radius;
You should update your software

**3mf File for This Bug**
any 3mf I saved from bambu studio

**To Reproduce**
save a 3mf from bambu and open it in Orca

**Expected behavior**
for the Slicer to be updated

**Screenshots**
[image]

**Printer model**
Bambu Labs X1C

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 22H2

neelyrock, 2023-06-16:
Can confirm I am also seeing this on Mac OS (Ventura 13.4).

fourtwizzy, 2023-07-12:
@SoftFever Low hanging fruit. This was resolved in the nightly builds complements of commit [bdf8c8d](https://github.com/SoftFever/OrcaSlicer/commit/bdf8c8d5b1b1ef87c392f54124371cdfed79650f).
Obviously it hasn't been pushed to the latest release, but easy to track and close :)

closed 2023-11-21 by github-actions[bot]