OrcaSlicer/OrcaSlicer #1350: Recent Projects not using custom saved presets

Depep1, 2023-06-15:
**Describe the bug**
After saving a recent project, closing, and reopening, the project does not use any custom presets. Instead, it modifies system presets. But even the modifications are different than what is saved in the custom preset. 

**To Reproduce**
For my system. Create new project. Slice with custom presets. Save as 3mf. Close Orca, Open Orca. Open 3mf project. the project doesn't use custom presets. 

[image]

Ender 3 v2
**Desktop (please complete the following information):**
 - OS: windows 10
 - Version 22H2

Depep1, 2023-06-15:
[PI3MK3M_03_Escape_Fork_Small_Clearance_2.zip](https://github.com/SoftFever/OrcaSlicer/files/11759779/PI3MK3M_03_Escape_Fork_Small_Clearance_2.zip)
After converting the 3mf to a zip and looking in the project_settings.config file, the output is almost correct. Filament and print setting id are correct but printer_setting_id is not correct. It should just be called "0.4mm Nozzle". 

    "filament_settings_id": [
        "Inland Pla Pro"
    ],
"print_settings_id": "0.20mm Ender3V2 (0.40mm Nozzle)",

"printer_settings_id": "Creality Ender-3 V2 0.4 nozzle",

Alexx40in, 2023-06-18:
i have the same problem
Before 
[image]
After reopening
[image]

closed 2023-11-21 by github-actions[bot]