OrcaSlicer/OrcaSlicer #1351: Fusion360 export stl using external application.

favoritewky, 2023-06-15:
**Is your feature request related to a problem? Please describe.**
Fusion360 export stl using external application like *PRUSA* can work good.

**Describe the solution you'd like**
But the orcaslicer can open a stl as the exe paramerter.

**Describe alternatives you've considered**
Can you make it work?

**Additional context**
Add any other context or screenshots about the feature request here.

julie777, 2023-06-19:
This should be considered as part of the single instance option. 
When the application is started by Fusion 360 it uses a command line `.../path/OrcaSlicer.exe STL_file_path_from_F360`.

This is also what happens when you drag a file to the slicer window, or when you use "open with" or have the file extension bound to the application.

Orca/Studio do not handle any of these well on Windows 10.

favoritewky, 2023-06-20:
> This should be considered as part of the single instance option. When the application is started by Fusion 360 it uses a command line `.../path/OrcaSlicer.exe STL_file_path_from_F360`.
>
> This is also [...]

Yes， *PRUSA* can handle this well so can you implement this please?

closed 2023-11-21 by github-actions[bot]