OrcaSlicer/OrcaSlicer #1353: Raise3D Pro2 direct gCode upload

Internet3xplorer, 2023-06-15:
**Is your feature request related to a problem? Please describe.**
I don't want to use Ideamaker or an USB Stick to upload gCode to the Raise3D Pro2 printer.

**Describe the solution you'd like**
I want to connect to the Raise3D Pro2 directly in OrcaSlicer to upload the gCode files through the local network.

**Describe alternatives you've considered**
The only way to upload gCode files sliced in OrcaSlicer to the Raise3D Pro2 is to use a USB Stick or Ideamaker to upload the files.

**Additional context**
Add any other context or screenshots about the feature request here.

kylek29, 2023-06-15:
I'm not sure that'd be doable since it appears Raise3d uses IdeaMaker and its own proprietary cloud. We only have BambuLab's cloud in OrcaSlicer because it's part of the fork and they didn't disable the API's for non-BambuLab software.

Does the printer open an FTP or file share? That's how I used to do it with a few printers -- you map a network share and then slice to GCode file, saving directly into the share.

fr3ak2402, 2023-06-15:
I'm a little confused right now.... Don't you mean Raise3D Pro3?

The profiles of the Raise3D Pro2 & Pro2 Plus have only recently been completed and will only be included in the next version of the OrcaSlicer.

Since I had only created the profiles for the Raise3D Pro3 & Pro3 Plus for the time being and had released them for version V1.6.3

So if the Raise3D Pro2 should be accessible via a local network, then you should already be able to access this printer, unless it uses a cloud interface like the Bambulab.

Then it will be difficult to send a GCode directly from the OrcaSlicer to the printer.
To send a file anyway, you can use an OctoPrint or Repetier server to send your file from the slicer to the server. Then you only have to start printing via the server.

However, if the printer does not use a cloud interface and its firmware has the option of receiving files via its network interface, you need to know the host type, the suitable port and the IP address of the printer in your network...

Internet3xplorer, 2023-06-15:
Thank you for your reply. I am using the Raise3D Pro2, but this printer should be very similar to the Pro3. Is there a guide to connect the Pro3 in OrcaSllicer? Ideamaker should send the files via Network to the printer, but i can't find a way to successfully connect to the printer in OrcaSlicer.

bani6809, 2023-10-04:
at least my raise3d n2 uploads directly to the printer, there is no cloud.

Internet3xplorer, 2023-10-05:
> at least my raise3d n2 uploads directly to the printer, there is no cloud. ideamaker connects directly to the printer.

You can upload form Orcaslicer directly to your Raise3D printer? What settings did you use?

closed 2024-01-11 by github-actions[bot]