OrcaSlicer/OrcaSlicer #1354: Items in the "Print plate" roll down do nothing

fleutot, 2023-06-15:
**Describe the bug**
After slicing, I click on the down arrow beside the green button "Print plate". I then choose one option, for example "Export G-code file".

Nothing happens.

**Desktop (please complete the following information):**
 - Ubuntu 22.10, which uses gtk4. In window manager xmonad, which is non-reparenting.
 - OrcaSlicer 1.6.3

julie777, 2023-06-19:
@fleutot 
It is a terrible interface. When you select an item in the drop down, instead of being a menu item that does something like you expect, it is actually changing what the button does. The label in the Print Plate button should change to Export G-Code and then you have to click the button.

fleutot, 2023-11-06:
I really wish that UI devs would refrain from doing their own UI design. I'm having loads of troubles with custom dropdowns, transparency, and what not. Right now I cannot use Bambu/OrcaSlicer in full screen, the dropdown menus don't show up. 

I wish for an option that makes the UI only use standard GTK elements. That would also be more respectful of the choices I make for my OS interface, and keep things consistent.

fleutot, 2023-11-06:
I mean, this is cleay broken:

https://github.com/SoftFever/OrcaSlicer/assets/178409/176f0133-118d-4566-8735-871ffcb6daac

closed 2024-02-19 by github-actions[bot]