OrcaSlicer/OrcaSlicer #1358: Failing to understand the code generated by toolchange

thisiscam, 2023-06-15:
Hi,

Thank you for this wonderful slicer. 
I am having difficulty understanding how orcaslicer generates gcode for a toolchange. 
For example, consider this piece of gcode generated, with the following options:
Machine change filament gcode: `CHANGE_FILAMENT_GCODE ` 
- Filament start gcode: `FILAMENT_START_GCODE `
- Filament end gcode: `FILAMENT_END_GCODE `
- Prime tower: enabled

```
...
G1 X135.94 Y166.425 E.01504
G1 X136.502 Y165.863 E.02365
; stop printing object test.stl id:0 copy 0
EXCLUDE_OBJECT_END NAME=Cowl_NoProbe.stl_id_0_copy_0
;HEIGHT:0.200000
;TYPE:Prime tower
;WIDTH:0.500000
;--------------------
; CP TOOLCHANGE START
; toolchange #1
; material : ASA -> ASA
;--------------------
M220 S100
FILAMENT_END_GCODE

G10 ; retract
CHANGE_FILAMENT_GCODE
M106 S0
G1 X82.765 Y208.753 F21000
G1 Z.6
G1 Z.4
G11 ; unretract
T2

FILAMENT_START_GCODE

G4 S0
G1  X82.265 Y208.753 
G1  Y208.253 
G1  X117.265  E1.3302
G1  Y228.753  E0.7791
G1  X82.265  E1.3302
G1  Y208.253  E0.7791
G1  X82.765 Y208.753  
; CP TOOLCHANGE WIPE
G1  X116.890  E1.2970 F1584
G1  Y210.086  E0.0507
...
```

I am having trouble understanding what the slicer is doing between the `FILAMENT_END_GCODE ` and `T2`:
```
G10 ; retract
CHANGE_FILAMENT_GCODE
M106 S0
G1 X82.765 Y208.753 F21000
G1 Z.6
G1 Z.4
G11 ; unretract
```

It seems to be moving to a prime tower? But then why is it doing 
```
G1 Z.6
G1 Z.4
```?

It almost seems like the `G1 Z.6` is intended as a z lift and should come before the line `G1 X82.765 Y208.753 F21000` (move to prime region)?
Further, is there anyway I can overwrite the `T2` to something more custom? Ideally, there should be placeholders for both pre-toolchange (the currently `change filament gcode` that happens prior to traveling to the prime region), the actual tool change (the currently `T2`), and possibly a post-toolchange (that happens after the movement in the prime tower).

thisiscam, 2023-06-16:
After digging at the [source code](https://github.com/SoftFever/OrcaSlicer/blob/4a799aa7705d14767c7de5a7ebb7d1ce61afa0cc/src/libslic3r/GCode.cpp#L561) a bit, I realized that the slicer expects the "filament change g-code" section to contain at least one uncommented `T[next_extruder]`. This is kind of surprising since my tool change is actually invoked through a custom klipper macro with a different calling convention.

 I have a workaround of "double changing" right now that looks like this:
```
DO_TOOL_CHANGE TOOL=[next_extruder] ..some other args..
T[next_extruder]  ; a no-op to make orcaslicer happy 
```

But, I'm still not sure what's happening with the double-z moves (that feels like a bug to me). 
And it would be nice if Orca can have more placeholder hooks into the toolchange process, and make things more expilicit (rather than relying on "the user must define T[next_extruder]"

jonathanfrem, 2023-07-16:
Also experiencing this issue. Workaround for removing the T2 in superslicer doesnt seem to work in orcaslicer.

This is what I've been using to avoid the first toolchange in SS `{if layer_z > -1}T[next_extruder]{endif}`

closed 2023-11-21 by github-actions[bot]