OrcaSlicer/OrcaSlicer #1360: Inconstant flow

emlnpro, 2023-06-16:
**Describe the bug**
On orca slicer, same layer can have different flow without any reason, wich is not good for consistency.
Same profile, the filament minimum layer time box uncheck same result

**3mf File for This Bug**
[Soutient plafond.zip](https://github.com/SoftFever/OrcaSlicer/files/11767876/Soutient.plafond.zip)

**To Reproduce**
0.16mm layer height

**Expected behavior**

**Screenshots**
First Bambu Sliccer
[image]

Then Orca slicer
[image]

**Printer model**
X1C 

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version : 1.6.3

julie777, 2023-06-19:
@emlnpro 
I have noticed this with some of my models. If I remember correctly it was caused by something in the model geometry changing layer time. There might have been a small feature starting at that layer, or maybe something in the infill changed.

Does this have to do with layer time causing speed changes?
Or max volumetric flow rate limiting speed for some layers?

I just realized that your images are showing flow rate for BS and Speed for Orca. They are not comparable.

julie777, 2023-06-19:
@emlnpro 
For some reason when I load your project into either Bambu Studio or Orca Slicer none of the settings are set correctly. It seems to only have the STL in it. 

To actually see if I can reproduce your results I need to know all the settings, including filament, especially if you have a tuned filament in Orca.

hatkyinc2, 2023-11-18:
you should probably cross reference the speed to the flow
the speed changes because of multiple things,
roughly 
line type + geometry + acceleration = speed. (can be slowed by other parameters)
speed * layer height * extrusion with = flow

closed 2024-02-25 by github-actions[bot]