OrcaSlicer/OrcaSlicer #1366: Incorrect wall order when inner-outer-inner/infill and 2 walls selected

alexrmay91, 2023-06-16:
**Describe the bug**
I set `Order of inner wall/outer wall/infill` to 'inner-outer-inner/infill' for almost all prints. This functions correctly unless `Wall loops` is set to `2`. When both of these are set, inside-loops (i.e. holes) print outer-inner while outside loops (i.e. outside of the object) print outer-inner. See the example below where the outside wall is printing first on the inner loop, opposite on the outer loop.

**3mf File for This Bug**
All models with any type of hole inside the part.

**To Reproduce**
Use the settings listed above. Select `3` walls or `inner/outer/infill` for normal behavior.

**Expected behavior**
When set to `inner-outer-inner/infill` and `2` walls, it should print with the inner wall first, especially on inside holes. This behavior leads to print failures, especially on models with countersunk holes (or other internal overhang). The walls tend to not stick to the previous layer and end up crossing the middle of the hole.

**Screenshots**

[image]

[image]

**Printer model**
Voron Trident

**Desktop (please complete the following information):**
 - Windows
 - V1.6.3

igiannakas, 2023-06-17:
This is not a bug, it is supposed to work this way. The inner outer inner by default needs 3 perimeters to work - the first inner is the third wall, then it prints the outer perimeter then the inner perimeter next to the wall. 

This is meant to work like the outer perimeter first mode but rather print an internal wall first away from the outer wall in order to improve the artefacts on the z seam that may occur from using outer perimeter first.

alexrmay91, 2023-06-17:
I understand how it works. I think my original posts may not have been clear?

I leave the wall order to inner-outer-inner by default, but sometimes print with 2 walls. When that happens, the walls on inside hole features print in a different order than walls on outside features. Look at the photos/toolhead. Picture 1, inside feature, orange is being layed down first. Picture 2, outside feature, yellow before orange. These pictures are using the same settings, no re-slicing.

igiannakas, 2023-06-17:
I’ve fixed this issue in my PR 1352. In this Pr outside walls are favoured over inside.

If you can upload the stl I’ll check if it happens with the updated version I’ve submitted.

alexrmay91, 2023-06-17:
The model in the screenshots is from [here on printables](https://www.printables.com/model/1161-phonetablet-stand-flat-fold-print-in-place/files), the small version.

igiannakas, 2023-06-17:
> The model in the screenshots is from [here on printables](https://www.printables.com/model/1161-phonetablet-stand-flat-fold-print-in-place/files), the small version.

With the PR #1352 it is printing consistently with outer wall first when only 2 walls are selected as it falls back to the outer perimeter first logic.

[image]
[image]

When three walls are selected it prints the innermost perimeter correctly, then the outermost.
[image]

alexrmay91, 2023-06-17:
I see. Sounds like I'll just have to remember to turn it back to inner-outer when I use two walls. Thanks for following up.

closed 2023-11-21 by github-actions[bot]