OrcaSlicer/OrcaSlicer #1377: Change filament issue

Alexx40in, 2023-06-18:
when I add a filament change command on a certain layer, another m600 command is added to the beginning of the file.
Since I use manual filament change, the printer pauses at the start of printing.

Alexx40in, 2023-06-20:
> you can use any filament and temp but only one nozzle/filament diameter pls add a second to process and change the parameter. but nozzle diameter is in printer settings

I don't understand very well what you are asking me to do.

When adding a filament change command, PrusaSlicer inserts the m600 command
Orca slicer adds extruder change command T0 to start and filament change position T1
This does not take into account the fact that the printer does not have a second extruder.

hyprodimus, 2023-06-25:
I found this bug too. When you add something like M600 to the Change Filament G-code section, it will add that command tot he start of the G-code which is pretty annoying.

bhast2, 2023-07-27:
I have the same problem when I add the m600 code to the machine gcode it puts an m600 at the begin of the print and it tries to changes the filament after the purge line is printed
How can this be fixed??

Bazeone, 2023-08-16:
I have the same issue and would like not to have to manually edit the file and remove the first M600 command.

l8nite, 2023-09-17:
+1 on this issue, ran into it yesterday.

Luigisvc, 2023-09-25:
Same issue here can I open a BUG report?, as appears no one from DEV noticed this topic here

sguth-affirm, 2023-09-25:
@SoftFever fyi

bpkenn, 2023-09-26:
I had the same issue today

wmariz, 2023-10-06:
Workaround: https://github.com/SoftFever/OrcaSlicer/issues/2230#issuecomment-1750086127

SoftFever, 2023-10-06:
Hey guys,

I can't repro the issue with the latest dev build.
Can you attach a project file?

Bazeone, 2023-10-06:
> Hey guys,
> 
> I can't repro the issue with the latest dev build. Can you attach a project file?

[color_change_bin.zip](https://github.com/SoftFever/OrcaSlicer/files/12829390/color_change_bin.zip)

bpkenn, 2023-10-06:
I reproduced the pause at the start by:

- Set Printer settings -> Machine G-Code -> Change filament G-code to value of "M600" or even "PAUSE"
- Add second filament
- change object color 2nd filament

I am assuming that since I'm manually swapping the filament, i already have the 2nd filament color loaded and ready to print, but Orca is assuming AMS type setup.  Is it possible to have a "manual MMU override" that will just run the change filament gcode only at the layers we add a filament change?

Ocraftyone, 2023-10-09:
> Is it possible to have a "manual MMU override" that will just run the change filament gcode only at the layers we add a filament change?

I am currently working on a feature like this. Should hopefully be complete if a few days.

closed 2023-10-22 by SoftFever

ToXIc69, 2024-01-28:
> I have the same problem when I add the m600 code to the machine gcode it puts an m600 at the begin of the print and it tries to changes the filament after the purge line is printed How can this be fixed??

exactly my issue.. after the purge line (in my start macro) it runs my m600 which pauses and retracts filament.

Bazeone, 2024-02-15:
still an issue with version 1.9.0.
quick workaround:
{if layer_z != 0}M600{endif}
{if layer_z != 0}G1 E0.4 F1500 ; prime after color change{endif}
for a Prusa.

Luigisvc, 2024-02-16:
This bug is already solved, please check on your Orca 1.90 and mark this option below, this will correct the behavior for printers without multimaterial feeders. 

[image]

Also put a M600 command in the Machine Gcode . Or another command if you use klipper 

[image]

ToXIc69, 2024-02-17:
thanks ill try it out

Naahuel, 2024-04-10:
> This bug is already solved, please check on your Orca 1.90 and mark this option below, this will correct the behavior for printers without multimaterial feeders. [...]

This fixed it for me on Orcaslicer 2.0.0, it drove me nuts. I had to manually remove the m600 command at the beginning, now it works as it should.

PR #2379 (2023-10-22): Add "Manual filament change" option and logic
This option does a few things that make a SEMM setup where you are manually swapping the filament on pauses a bit easier

1) disables calling the "on filament change" gcode during setup. This may be useful on a machine that can change the filament itself, but for manually changing the filament, it is an unnecessary pause. It should be assumed the first inserted filament is the first one to be used.

2) tool change command is changed to a gcode tag. This prevents errors on machines that do not understand the command possibly resulting in a failed pr [... cut]