OrcaSlicer/OrcaSlicer #1378: Dowels changing scale/position/orientation

mitsuracer85, 2023-06-18:
**Describe the bug**
While performing a cut with dowels, the negative dowel shape is twice as large as the dowel itself, and the position is not oriented on either of the the parts properly, both in position and angle.

**How to reproduce bug**
select cut tool
select add connectors
select dowel
place dowel positions
confirm connectors
select keep top and bottom parts
select place on cut for both parts
perform cut

**Expected behavior**
a negative cylinder should be placed in both parts at the positions previously selected.

**Screenshots**
[image]
[image]

**Printer model**
X1C

**Desktop (please complete the following information):**
 - OS: Windows 10

mrmccain9, 2023-06-25:
Same issue here on intel Mac version when printing a part at 250% scale.

One hopefully helpful data point is that I don't have the issue if I cut the dowel holes before scaling from 100% to 250% (but then the clearances are too tight and the dowels don't fit).

If I cut the dowel holes after scaling from 100% to 250%, the holes get placed completely outside of the part and in a random orientation.

mitsuracer85, 2023-06-25:
Right, I first noticed it after I scaled this file to 200%, so I decided to try it at 100% and then scaling up but it was still doing it. It works ok with pegs but not dowels. Although I don't have this issue on other files...this is the only file that's having this issue for me so far for some reason.

Dustmuffins, 2023-07-14:
Yes, I've noticed this bug as well. If you add the dowels, then scale the part, there is no issue. If you scale the part first then add the dowels, the size gets messed up.

You can replicate this by creating a cube and scaling it to 200%, then adding dowels. The negative cylinders will be in the complete wrong location and size will be wrong.

closed 2023-11-21 by github-actions[bot]