OrcaSlicer/OrcaSlicer #1394: Exclude object on Bambu

rkarlsba, 2023-06-21:
I own a Bambu X-1 Carbon with an AMS and it'd be very nice to see exclude object or similar in OrcaSlicer. Is this possible to implement?

abstractFlo, 2023-06-28:
exclude objects has nothing to do with the slicer the slicer only labled the objects and the printer firmware can evaluate this and make some exclusions.

The Bambu Printer does not have this feature like any other popular klipper based printer

kylek29, 2023-06-30:
As abstractFlo said, that needs a firmware change on the X-1C printers. It's a feature I've wanted to (to be able to select per-object reprints). The way it'd work (I'd assume) is that the Slicer would drop a hint in the GCode to segment the file between objects (I *think* it already has this ability or does this anyways) and the firmware would handle the rest.

I'd suggest making the suggestion here: https://forum.bambulab.com/

Schokobecher, 2023-07-20:
This is now available in [BambuStudio v1.7.0](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.00.53).
The printer needs to be on FW V01.06

Schokobecher, 2023-08-02:
The mentioned firmware and BambuStudio version is now out of Beta

rkarlsba, 2023-08-02:
Any idea how to update the printer?
[image]

rkarlsba, 2023-08-02:
Never mind - I rebooted the printer and it found it.

rkarlsba, 2023-08-02:
But I can't find a way to cancel an object. Nothing in the GUI - nothing I can find anywhere.

Schokobecher, 2023-08-14:
I'm pretty sure the current version of OrcaSlicer hasn't back ported the feature yet

drewrox2009, 2023-08-14:
first time i needed this feature, it notifies me that i need a newer version of bambu slicer.

rkarlsba, 2023-09-14:
AFAICT, there's no way to ignore/exclude objects from either BambuStudio nor OrcaSlicer. The possibility is there, though, on the printer console and in BambuHandy. This would be very nice to get into OrcaSlicer, since I obviously need it sometimes (like just now when things started to peel off th ebuildplate at layer five or something - just dirty buildplate, but some of those 12 objects could have been saved).

closed 2023-12-21 by github-actions[bot]

rkarlsba, 2025-02-03:
This must be reopnened and fixed. It doesn't fix anything that the bug report is old.