OrcaSlicer/OrcaSlicer #1396: More Advanced Internal Solid Infill Control

ruredi87, 2023-06-22:
**Is your feature request related to a problem? Please describe.**
Yes. Orcaslicer adds unnecessary internal solid infill near walls. Infill prints last in order, so it does not needed near sloping walls. It causing waste of time and material. Some times slicer creating internal infill using rectelinear pattern. This causes even more time to print.

**Describe the solution you'd like**
Superslicer solves it with "NO SOLID INFILL OVER : X WALLS" option. This option causes to more advanced control of internal solid infill. If we use value of "1" slicer does not add solid infill over 1 perimeter. This solves problem.

[image]
[image]
[image]

[Cat_bowl_deep internal infill test.zip](https://github.com/SoftFever/OrcaSlicer/files/11827363/Cat_bowl_deep.internal.infill.test.zip)

tome9111991, 2023-06-22:
You have tried to disable infill_anchor ?

ruredi87, 2023-06-22:
> You have tried to disable infill_anchor ?

I tried, it does not work. Only way to prevent from it to disable infill completely .

closed 2023-11-21 by github-actions[bot]

rlabusiness, 2024-01-01:
I believe I'm having surface quality issues caused by this as well. The extra solid infill printed over top of sparse infill when the outer surface is a slope appears to build up one layer at a time and cause inconsistencies in the surface/outer wall (photo attached).

[image]

mstrblueskys, 2024-09-07:
> > You have tried to disable infill_anchor ?
> 
> I tried, it does not work. Only way to prevent from it to disable infill completely .
> 
> You can see on superlicer screenshot. Infill anchor is enabled but ss doesnot generate internal solid infill which respects to "NO SOLID INFILL OVER : X WALLS"

How do you disable solid infill completely?