OrcaSlicer/OrcaSlicer #1399: Spiral vase mode for objects

Alexx40in, 2023-06-22:
I really like the ability to place my project on several plates

However, if the project has a model that prints in spiral vase mode, I can't include it in the project
Please add the ability to add a spiral vase mode to individual objects with a condition: that this is the only object on the plate

And one more wish. I would like to apply the settings not only to individual objects, but also to a separate plate as a whole

Dynom, 2023-10-31:
Upvoted. Currently the workaround is creating two projects, one with vase mode and one without. Which works, but the multi-plate system works so well for multi-part projects, that it would be fantastic to have this as a per-plate option.

LupusTheCanine, 2024-02-01:
That would be great.

closed 2024-05-19 by github-actions[bot]