OrcaSlicer/OrcaSlicer #1400: [FR] Disable Thumbnail

foreachthing, 2023-06-22:
#### Describe the solution you'd like
I'd like to be able to disable the thumbnail generator. Since our printer does not support it and we only print from SD-card, we don't need thumbnails.

#### Describe alternatives you've considered
* Manually removed the thumbnail code from gcode files.
* changed thumbnails size to 0x0 -> Error.
* changed thumbnails size to 1x1 -> still produces code.

#### Printer
* Ultimaker 2

cron410, 2023-06-28:
[image]

This seems to work for me. 

When I export the gcode, there's no encoded image within the file, the whole THUMBNAIL_BLOCK is gone.

foreachthing, 2023-06-28:
@cron410 , ok!!! Now I feel silly ... because that's what I tried first. But OrcaSliced did not let me remove this value. Now it works ...
Thanks!

closed 2023-06-28 by foreachthing