OrcaSlicer/OrcaSlicer #1404: Choose ironing layers

freakybuckey, 2023-06-22:
Sometimes I'd like to iron both the first layer (in order to remove any artifacts where adjacent lines meet) and the top layer, but the only way to do this at present is to select ironing for "All solid layers". This ends up causing the shell layers above the bottom layer to also be ironed, adding unnecessary print time. 
 
The ability to choose specific layers to iron would solve this problem.  An option to iron only the topmost and bottommost surfaces would also work for my specific problem but be less functional overall.

pemerick07, 2023-06-23:
In V1.6.3 Height Range Modifier has been added. You can use HRM to selectively specify where to iron. I've never tried ironing the first layer, but HRM worked for me to only iron flange and o-ring faces on parts that have sealing features. Specify a height range and select iron all top surfaces, which will iron all surfaces within the height range.

closed 2023-11-21 by github-actions[bot]

megothss, 2023-11-28:
+1 for this feature. It would be really nice to have the option to iron the first layer for better quality at the bottom of the print.