OrcaSlicer/OrcaSlicer #1405: Corrected Czech translation

renemosner, 2023-06-22:
I am sending the completed Czech translation in the attachment. Modified original version.
Thanks for your work.
Sincerely, René Mošner

[OrcaSlicer_cs.zip](https://github.com/SoftFever/OrcaSlicer/files/11839565/OrcaSlicer_cs.zip)

fr3ak2402, 2023-06-22:
Thank you for your work, I appreciate it very much.
I will forward this...

renemosner, 2023-06-23:
Is it possible to send me the current *.pot file before new versions of OrcaSlicer are released?  So that I can update the translations before release. 
So that the new version is always up to date for cs localization? 
Only if possible. 
Thank you with regards and thanks René Mošner

renemosner, 2023-07-17:
@SoftFever

Hello, will the updated cs/cz translation be used?  thank you for answer

SoftFever, 2023-07-17:
> @SoftFever
> 
> Hello, will the updated cs/cz translation be used? thank you for answer

Ah, sorry. I overlooked this thread.
This one is update to date: https://github.com/SoftFever/OrcaSlicer/blob/main/bbl/i18n/OrcaSlicer.pot

Free feel to create a PR.
If you need help, just tag me, I can create a PR on behalf of you(assign you as the author)

renemosner, 2023-07-17:
@SoftFever

Here is the current translation according to the latest *.pot file, please PR.
Thank you very much for your great work.
Best regards, Mošner.

[OrcaSlicer_cs.zip](https://github.com/SoftFever/OrcaSlicer/files/12073623/OrcaSlicer_cs.zip)

renemosner, 2023-08-05:
@SoftFever

Can I ask for a PR for the above translation edit?  Thank you Mošner

SoftFever, 2023-08-05:
> @SoftFever
> 
> Can I ask for a PR for the above translation edit? Thank you Mošner

Done, can you review changes? https://github.com/SoftFever/OrcaSlicer/pull/1702

Let me know in case I misread your word. I can delete the PR if you want to create the PR by yourself.
I

renemosner, 2023-08-05:
> > @SoftFever
> > Can I ask for a PR for the above translation edit? Thank you Mošner
> 
> Done, can you review changes? #1702
> 
> Let me know in case I misread your word. I can delete the PR if you want to create the PR by yourself. I

Thank you very much.

closed 2023-09-05 by renemosner