OrcaSlicer/OrcaSlicer #1409: Scroll in Preview moves 2 layers at a time

sdooweloc, 2023-06-23:
**Describe the bug**
When scrolling on the layer height slector bar in the Preview window after slicing the bar jumps 2 layers at a time making some layers impossible or annoying to get to by scrolling and causing some inconvenience when trying to set up layer change insertions.

I am using a physical mouse scroll wheel, not a trackpad or anything that could cause discrepancies when scrolling and I know from games, other slicers, and general use that this mouse scroll is interpreted as 1 line of scroll by my machine elsewhere. 

**3mf File for This Bug**
Unable to attach 3mf as github does not support it as a file type and re-typing files risks data loss/corruption but this happens to every model/file etc I run through the slicer.

**To Reproduce**
Add items to build plate
slice
Try to scroll over the layer indicator bar.

**Expected behavior**
Scrolling only moves 1 layer at a time

**Screenshots**
[image]
[image]

**Printer model**
Voron 2.4 300mm

**Desktop (please complete the following information):**
 - OS: ArchLinux 
 - Version 1.6.3

kmarty, 2023-10-04:
I'm experiencing the same issue.

- OS: Debian 12
- Orcaslicer version: 1.7.0

closed 2024-01-10 by github-actions[bot]

kmarty, 2024-01-10:
“completed” :-/. The problem persists

kmarty, 2024-01-14:
For reasons unknown to me, the `m_canvas->Bind(wxEVT_MOUSEWHEEL, &GLCanvas3D::on_mouse_wheel, this);` is triggered twice every click of mouse wheel.
This issue seems to be inherited (at least) from Superslicer (didn't tried it in Prusaslicer, but I believe it is there too, since this part is basically unchanged), but since `GLCanvas3D::on_mouse_wheel()` is used only for zooming (DoubleSlider used in Superslicer and Prusaslicer has own separate `wxEVT_MOUSEWHEEL` binding which works fine - it is triggered just once per click), it somehow isn't exposed (yet, the `GLCanvas3D::on_mouse_wheel()` used there (for zooming) is triggered twice too, but zoom reacts only once).

In Orcaslicer, they used this function also for IMSlider's `m_layers_slider` and `m_moves_slider` and since this function is triggered twice per click, it is also called twice.

I'm not C++ nor wxMouseEvent coder so the only way how I "solve" it was ugly hack skipping every second trigger (surprisingly, the zooming still reacts as it should):
```
diff --git a/src/slic3r/GUI/GLCanvas3D.cpp b/src/slic3r/GUI/GLCanvas3D.cpp
index ae48ff8eb..ff7149c1b 100644
--- a/src/slic3r/GUI/GLCanvas3D.cpp
+++ b/src/slic3r/GUI/GLCanvas3D.cpp
@@ -3598,8 +3598,19 @@ void GLCanvas3D::on_key(wxKeyEvent& evt)
     }
 }
 
+#ifdef __LINUX__
+static bool slider_hack_against_wheel_doubletrigger = false;
+#endif
+
 void GLCanvas3D::on_mouse_wheel(wxMouseEvent& evt)
 {
+#ifdef __LINUX__
+    // Ugly, nasty hack
+    slider_hack_against_wheel_doubletrigger = slider_hack_against_wheel_doubletrigger ? false : true;
+    if (!slider_hack_against_wheel_doubletrigger)
+           return;
+#endif
+
 #ifdef WIN32
     // Try to filter out spurious mouse wheel events comming from 3D mouse.
     if (wxGetApp().plater()->get_mouse3d_controller().process_mouse_wheel())
```
(it works to me, but definitely not for production release)

sdooweloc, 2024-01-14:
Yeah that definitely looks hacky but maybe it can give @SoftFever something to go off of for a cleaner fix knowing the code better. Though you mentioned the same code seems to be in SuperSlicer though I dont experience the issue in there so not sure where it may trace back to for the root cause and best place to fix.

kmarty, 2024-01-14:
The code in SuperSlicer is same (except the added parts with IMSlider) and it is triggered twice too! But somehow the zoom skip the duplicity so you'll se just one click for the zoom (as well as zoom in Orcaslicer).