OrcaSlicer/OrcaSlicer #1411: BIQU B1 Stock profile not slicing/GCODE Bug

Lian556, 2023-06-23:
Biqu B1 profile is not slicing, it gives:

> Failed to generate G-code for invalid custom G-code.
> machine_start_gcode Parsing error at line 3: Not a variable name
> M140 S{material_bed_temperature_layer_0}; Set Heat Bed temperature
>   
> Please check the custom G-code or use the default custom G-code.

[image]

OS: Windows 11 Professional X64
OrcaSlicer: 1.6.3

cron410, 2023-06-28:
@Duncro fixed it and shared the B1 profile. 

[BIQU-B1-0.4-nozzle.json.txt](https://github.com/SoftFever/OrcaSlicer/files/11888295/BIQU-B1-0.4-nozzle.json.txt)

You would replace your B1 profile json file found at this path on Windows:
C:\Users___________\AppData\Roaming\OrcaSlicer\user\default\machine

Remove the .txt extension as well.

hho, 2023-06-28:
Thanks for the fix, but I think it's not entirely correct: It uses the bed temp and extruder temp for the other layers, not for the first layer in the start gcode. And the end gcode doesn't use the bed dimensions (but a hardcoded 220mm).

I use for start:

```gcode
; BIQU B1 Start G-code
M117 Getting the bed up to temp!
M140 S{hot_plate_temp_initial_layer}        ; Set Heat Bed temperature
M190 S{hot_plate_temp_initial_layer}        ; Wait for Heat Bed temperature
M117 Getting the extruder up to temp!
M104 S{nozzle_temperature_initial_layer[0]} ; Set Extruder temperature
G92 E0; Reset Extruder
M117 Homing axes
G28; Home all axes
M109 S{nozzle_temperature_initial_layer[0]} ; Wait for Extruder temperature
G1 Z2.0 F3000                               ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X4.1 Y20 Z0.3 F5000.0                    ; Move to start position
M117 Purging
G1 X4.1 Y200.0 Z0.3 F1500.0 E15             ; Draw the first line
G1 X4.4 Y200.0 Z0.3 F5000.0                 ; Move to side a little
G1 X4.4 Y20 Z0.3 F1500.0 E30                ; Draw the second line
G92 E0                                      ; Reset Extruder
M117 Lets make
G1 Z2.0 F3000                               ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0                      ; Move over to prevent blob squish
```

And for end (referencing bed size):

```gcode
; BIQU B1 Default End Gcode
G91                                         ; Relative positioning
G1 E-2 F2700                                ; Retract a bit
G1 E-2 Z0.2 F2400                           ; Retract a bit more and raise Z
G1 X5 Y5 F3000                              ; Wipe out
G1 Z10                                      ; Raise Z by 10mm
G90                                         ; Return to absolute positioning
G1 X0 Y{print_bed_max[1]}                   ; Move bed forward all the way
M106 S0                                     ; Turn-off fan
M104 S0                                     ; Turn-off hotend
M140 S0                                     ; Turn-off bed
M84 X Y E                                   ; Disable all steppers but Z
```

hho, 2023-11-06:
Dear Bot, this issue is still present in 1.8.0-beta2.

closed 2024-02-19 by github-actions[bot]