OrcaSlicer/OrcaSlicer #1412: Auto Orient / Rotate model on import

iitsalex, 2023-06-23:
**Is your feature request related to a problem? Please describe.**
Superslicer has a feature to auto-orient / rotate models on import. It's very handy for those of us who use different infill angles than 45 degrees.

**Describe the solution you'd like**
A printer setting to set the angle at which models should be imported at.

**Describe alternatives you've considered**
No real alternatives besides manually rotating every single model I import (no bueno).

**Additional context**

[image]
[image]

tastyratz, 2023-08-09:
I would really love to see this as well! Especially since we can't perform what seems like ANY bulk or repeat operations on parts by selecting all at once from what I can tell.

iitsalex, 2023-08-11:
Started work on this. Have it working already. Just making some minor changes before putting it in CR.

closed 2023-11-21 by github-actions[bot]

iitsalex, 2023-12-14:
Just updated the CR for latest build.