OrcaSlicer/OrcaSlicer #1415: (OrcaSlicer 1.6.3) Travel speed very slow

UsSuggester, 2023-06-23:
I set the travel speed up to **200mm/s** but on every layer, I get a real travel speed of about **20mm/s.**

Anyway, on the graphics, slicer says that the travel speed is the one I set.

Question: Could this be a conflict between travel speed and "min print speed" on Part Cooling Fan? (It is the only speed parameter that I set with a very slow or similar value to the speed observed on traveling).

**Printer model**
Ender 3 S1 (on Marlin and Klipper)

**Desktop (please complete the following information):**
I'm on MacOS 13.4

blueeagle69, 2023-06-24:
I appear to have noticed the same issue.
I have my travel speed set to 180mm/s and my Initial layer travel speed set at 100%, but when it's printing, the nozzle is moving painfully slow. Including before the print starts. Prime line etc.

closed 2023-07-01 by UsSuggester

secretstanxxx, 2023-09-15:
how did you fix this

deemoss, 2023-10-17:
Has I get the same thing

Sate04, 2023-11-08:
was this ever solved? I'm having the same issue right now, I don't want to switch back to SuperSlicer but thats what I'm having to use right now.

deemoss, 2023-11-08:
> was this ever solved? I'm having the same issue right now, I don't want to switch back to SuperSlicer but thats what I'm having to use right now.

For me this was solved by creating a new printer profile from scratch.

Sate04, 2023-11-08:
Found the issue, OP was onto something with the spiral Z-Hop, I had copied it from the X1C slicer settings. Turning that off fixed my travels. Z-Hop is hidden in the printer profile extruder, making a new profile would've gotten rid of that, probably why it fixed the issue for you.

RTVRTV, 2024-01-04:
Under Filament settings --> Tab Cooling --> Turn off "Slow printing down for better layer cooling"

CodeMasterCody3D, 2024-02-23:
I know how to fix slow travel while spiral z hop. 
Recompile firmware, look for arc moves enable it, look for CNC_WORKSPACE. (ARC MOVES WILL BE ABOVE IT) I also enabled P moves or what ever it's called. I haven't had sleep been working on this for days. Your welcome. Spiral z hop for the win guys. Imma have a video out soon with a tutorial. Also I had to set jerk limits for I and J moves to 20 so it actually spins fast. Also my travel speed is 300mm/s ender 3 pro using Bambu hot end, bambu gcode for prime line and modified firmware for CNC workspace, arc moves, jerk is set to 20 for I and J moves (it's used for making the circle. P is also used in the gcode spiral z hop as well and I noticed it while scrolling the firmware. Like I said I'll have a video out soon. Sorry for vague information.