OrcaSlicer/OrcaSlicer #1418: Bed temp doesn't drop after first layer

Bakes70, 2023-06-24:
Hi there

Anyone else notice that their bed temp doesn't drop after the first layer. I am printing PLA with a 65degree first layer and 55degrees after the first layer. My nozzle temps drop from 210 to 205 as they should but I noticed that my bed stays at 65 for the full print.

I am using the latest Orca 1.6.3 version. This is only a small issue but maybe others have noticed it too. 

Thanks for creating great slicing software. Overall I am very happy with Orca slicer after moving over from Superslicer.

Noisyfox, 2023-07-23:
Is this the same as https://github.com/SoftFever/OrcaSlicer/discussions/1492?

Bakes70, 2023-07-24:
Seems to be similar to #1492. I found this doesn't always happen so perhaps this makes it difficult to debug.

Still a big fan of Orca Slicer regardless.

Noisyfox, 2023-07-24:
> I found this doesn't always happen so perhaps this makes it difficult to debug.

Yeah sadly I couldn't reproduce that one either.

closed 2023-11-21 by github-actions[bot]