OrcaSlicer/OrcaSlicer #1419: Zhop/Lift Z - options to select when it is active

jeremytodd1, 2023-06-24:
**Is your feature request related to a problem? Please describe.**
Disabling zhop entirely is a great way to eliminate stringing as well as shortening the print time by a considerable amount. The downside is that doing this will leave marks on top layers as the doesn't lift at all.

**Describe the solution you'd like**
Both SuperSlicer and PrusaSlicer have solutions that allows you to change when zhop is active.

SuperSlicer:
[image]

PrusaSlicer:
[image]

Currently SuperSlicer has an extra option to set it for "only top layers" which is perfect. Having that variable available would be amazing as well.

gbyleveldt, 2023-06-26:
I’ve reported this before - I hope it’ll be implemented soon. But I’m seeing other more important issues they’re trying to sort out

leandrolima-nyc, 2023-07-06:
The matter was previously addressed in issue #741, but no actions have been taken thus far.  You presented a more effective report, the issue should be merged rather than closing it outright.

closed 2023-11-21 by github-actions[bot]