OrcaSlicer/OrcaSlicer #1420: Unable to get a fast GCODE out for Creality K1

kaushikbm, 2023-06-24:
Because of the neat & clean interface I want Orca Slicer to work with. But unfortunately I'm unable to get a fast GCODE out of it. For example a model in Creality Print is giving me [09:41] (mm:ss) and the same one on Orca shows me [22:09] (mm:ss) after making a ditto profile. Any ideas ?

cron410, 2023-06-25:
What are your settings?

kaushikbm, 2023-06-25:
@cron410 Yes. I tried the GCODE. It takes almost the same amount of time as estimated by OrcaAs mentioned earlier, I copied all the settings from Creality Print.

davidja92, 2023-06-25:
i would double check acceleration and jerk speeds also see if theres a prints speed slowdown for cooling

kaushikbm, 2023-06-25:
@davidja92 I checked again and again, but it seems that Orca has got some limits set internally. I tried the same thing with Prusa slicer 2.6.0. It works better than Orca but still falls short. Maybe I think Creality Print has been tweaked for high speed performance.

cron410, 2023-06-25:
Not really. Have you looked at the volumetric flow in the printer profile?
Mine was defaulted to 12mm³

kaushikbm, 2023-06-25:
@cron410 Yes. I've set it to 32mm³ as per the printer's specs.

davidja92, 2023-06-25:
@kaushikbm last thing did you change the maximum printer motion abaility?
[image]

davidja92, 2023-06-25:
you can set your speeds to whatever you want but it will be limited by the motion ability settings

kaushikbm, 2023-06-26:
@davidja92 I copied all the motion ability settings from Creality Print. As mentioned earlier I checked all the possible settings many times.

Surion79, 2023-06-27:
Or Creality Print is using "custom" slicing algorithms, which are dedicated/specific to K1. At the top, someone suggested comparing the output gcode to view what kind of difference are there. you only checked settings in the program.

kaushikbm, 2023-06-28:
> @cron410 Yes. I tried the GCODE. It takes almost the same amount of time as estimated by Orca. As mentioned earlier, I copied all the settings from Creality Print.

@Surion79 I actually printed the GCODE and checked the timings of both. It's actually almost the same as calculated by the respective slicers.

rwmech, 2023-08-22:
Ive been having a very similar or the same problem.  I use the Bambu Slicer I get a file that has travel moves down to 43 minutes using the same acceleration 20000mm/s and a 500mm/s travel moves and Orca is reporting 1:30 minutes (Double) for the same model with a 600mm/s travel move.  Interesting for a similar lines of GCODE a G3 Move with F30000 feedrate on Bambu, and Orca is outputting a F36000 feedrate which makes sense given the increased speed but yet the calculated time is still double that of the bambu.  Not sure if its a calculation problem or truly making these moves slower. 

[image]
[image]

i9, 2023-09-08:
@kaushikbm have you looked at all related profiles/configs under Creality Print resources/sliceconfig/default/ folder?
Extruders, Machines, Materials, Profiles all have k1 related settings

rwmech, 2023-09-08:
I personally didn't go through the JSON, no.  I did validate in the UI all
configs matched.  I figured it could be an unexposed setting in JSON but
did not have the time to go to that length.  My work around was just to add
the K1 to Bambu slicer and use it from there.

kaushikbm, 2023-09-11:
> I personally didn't go through the JSON, no. I did validate in the UI all configs matched. I figured it could be an unexposed setting in JSON but did not have the time to go to that length. My work around was just to add the K1 to Bambu slicer and use it from there.
> […](#)

So are the GCODEs comparable as far as the timing goes ?

closed 2023-12-18 by github-actions[bot]