OrcaSlicer/OrcaSlicer #1423: Dremel 3D45 support

onehundredfeet, 2023-06-25:
**Is your feature request related to a problem? Please describe.**
I am unable to use my filament printer with your software, the Dremel 3D45.

**Describe the solution you'd like**
Support the Dremel 3D45 as a supported printer.

**Describe alternatives you've considered**
Other software.

**Additional context**
None.

closed 2023-11-21 by github-actions[bot]

cochcoder, 2024-04-30:
@onehundredfeet Do you still own this printer? If so, can you please test my PR #4777. Here's a [build](https://github.com/cochcoder/OrcaSlicer/actions/runs/8900135189) if you need it.

onehundredfeet, 2024-04-30:
I do.  It'll take me a couple days to get to it.  I'll test it and let you know.

cochcoder, 2024-04-30:
> I do. It'll take me a couple days to get to it. I'll test it and let you know.

That's great!

Klingler1998, 2024-05-26:
@cochcoder Hey how do I download that build? I could really use a better slicer for my 3d45 and orca looks great.

cochcoder, 2024-05-26:
@Klingler1998 You can download that latest build [here](https://github.com/SoftFever/OrcaSlicer/actions/runs/9240947622) by scrolling down and selecting the one for your platform. If you're on Windows, I highly recommend that you download `OrcaSlicer_Windows_V2.1.0-alpha_portable` until the official release is out. I recommend this as it'll give you the profiles you want without messing anything up with the version you currently have, just know that when you update the version you have installed and delete this portable version, you'll lose any data inside the portable version. 

If you can, please let me know if everything works and give me feedback as I don't own this machine.

cochcoder, 2024-05-27:
> @cochcoder Thank you! Looks like the same page you posted last month but the builds are clickable. It is probably for the best you don't own a Dremel printer, they are nothing but headaches lol.
> 
> 
> 
> I'll let you know how it works when I get home tonight to swap out prints.
> 
> 

Thanks, hopefully it somehow all works in one shot. Also, if you didn't already know this about the build pages, they all look the same but the files that are available to download are different due to changes in the code.

Klingler1998, 2024-05-27:
@cochcoder Alright so I have some feedback. The version does boot up and I was able to find the Dremel profiles. However, the same issues I experienced with stock Orca were present.

First and foremost, when loading my 3D45 with gcode sliced in orca, no extrusion occurs other than my manually entered purge line g-code. After the purge line is extruded the printer goes through the motions of printing without putting down any plastic.

The gcode in question:
G92 E0
G0 F4000
G0 X-115 ; set wipe start x
G0 Y-40 ; set wipe start y
G0 Z1
G0 X-115 Y40 F1500 E25 ; set wipe stop x and y
G92 E0

This is probably low on the list of priorities, but the build plate texture is disjointed. 
[image]

The printer itself does not seem to read the estimated print time from the gcode when loaded. IE the touch screen reads "not sure".

And lastly, and changes I make to build plate or nozzle temp seem to be ignored, With the printer resorting to 30 C on the plate, and 220 C on the nozzle.

These are identical to the issues which have kept me from using Orca originally. They were present when I made a custom profile for it.

**If I were to guess it is the need for proprietary Dremel gcode causing the issues. This addon for Cura fixed similar issues when I first switched off of print cloud. Your version of Orca is generating normal gcode. I cant say for sure if its the root cause, but I would be surprised if it wasn't.**

https://marketplace.ultimaker.com/app/cura/plugins/3PITechSolutions/DremelPrinterPlugin

Thank you again for your valiant attempt to clean up the mess of Dremel's programmers. <3

Hope this helps.

cochcoder, 2024-05-27:
Ok, I made several changes to the printer's gcode, can you test them? Also, do you see a thumbnail on the printer?

Start G-Code:

G90; use absolute coordinates
M83; extruder relative mode
M140 S[bed_temperature_initial_layer_single]; set final bed temp
M104 S150; set temporary nozzle temp to prevent oozing during homing
G28
G92 E0
M104 S[nozzle_temperature_initial_layer]; set final nozzle temp
M190 S[bed_temperature_initial_layer_single]; wait for bed temp to stabilize
M109 S[nozzle_temperature_initial_layer]; wait for nozzle temp to stabilize
G0 F4000 X-115 Y-40 Z1; move to position for purge line
G0 X-115 Y40 F1500 E25; purge line
G92 E0

End G-Code:

M104 S0; turn off nozzle
M140 S0; turn off bed
G92 E1; return print head to home
G1 E-1 F300
G162 Z F600
G162 X Y F2000
M84; disable stepper motors

Before layer change G-Code:

;BEFORE_LAYER_CHANGE; [layer_z]
G92 E0

Pause G-Code:

;@pause

Klingler1998, 2024-05-28:
Sure thing, next print is up in 1h 53m

cochcoder, 2024-05-29:
@Klingler1998, did you see any improvements?

cochcoder, 2024-05-29:
> @cochcoder Sorry got sidetracked.
> 
> Same issues. Also the print now starts at 150 C. It seems its not recognizing the other gcode temperature lines.
> 
> The whole no extrusion thing is really the only killer here.

I found a profile for PrusaSlicer that should work, I'll get the new build to you ASAP

cochcoder, 2024-05-29:
@Klingler1998 I have a build going [here](https://github.com/cochcoder/OrcaSlicer/actions/runs/9288870519), but you could also download [this](https://github.com/SoftFever/OrcaSlicer/files/15486714/Dremel.3D45.0.4.nozzle.json) and overwrite the current file at `/path-to-orca-slicer/resources/profiles/Dremel/machine/`. If you find any more issues, please let me know at #5496.

cochcoder, 2024-05-30:
> i am using also the Dremel 3d45 and downloaded your newest Version. The Printer should start in the middle of the printing plate, but he starts on the right corner and also wants to print next to the plate.
> 
> This is my test Print ![Unbenannt](https://private-user-images.githubusercontent.com/102697426/335369292-0acd3718-e98f-4d95-a02e-0b12a6176155.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcxMDc3MjQsIm5iZiI6MTcxNzEwNzQyNCwicGF0aCI6Ii8xMDI2OTc0MjYvMzM1MzY5MjkyLTBhY2QzNzE4LWU5OGYtNGQ5NS1hMDJlLTBiMTJhNjE3NjE1NS5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUzMFQyMjE3MDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05ZjhlYTRmNWMyNmNlOTRjOGIyMGNiMDU2MmFmMGM1ZWNmZTM1Zjg4Yjg0M2EyYTc3YTIwYzVlMzdiNTYxZTQ0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.xwRqvovANtSY_fXeJ76rC7atsDXslb32Ll8KJSNxna8)

Can you please move your print head to its origin and send me the coordinates of it?

cochcoder, 2024-05-31:
@Reevoy24 @Klingler1998 If one of you can send me some measurements in mm of the hot end, that would be great so that collisions can be prevented in the future. The measurements that are needed are: extruder clearance radius, extruder clearance height to the rod, and extruder clearance height to the lid.

cochcoder, 2024-06-06:
@Reevoy24 I don't mean to seem impatient, but do you have any updates?

Reevoy24, 2024-06-06:
> @Reevoy24 I don't mean to seem impatient, but do you have any updates?

My problem is that i cant send G-Code to my 3d45 so i cant give you the coordinates i tryed Simplify3D and Repetier Host with wifi and cable but it sends nothing back.... if you have any Idea how to get the data i would try my best.

cochcoder, 2024-06-06:
I made a custom G-Code that should home the printer and move the print head into four different corners. Please let me know which of these corners is closest to the head's origin and how close it is to it. Here is the custom [G-Code](https://github.com/user-attachments/files/15659545/find_origin.zip).

cochcoder, 2024-06-06:
> after Home it drives too far to the right and drives extremely into the wall and to the left too
> Same problem on the front right. Otherwise, Home was right in terms of position
> 
> I think the G1 is the most fitting one

Where did it go on the second move?

Reevoy24, 2024-06-06:
the second moved to the left front and bumped against the wall

Reevoy24, 2024-06-06:
I sliced a 1x1 cube perhaps the gcode helps you to fix it
[image]

[D3_Korper.txt](https://github.com/user-attachments/files/15667959/D3_Korper.txt)

cochcoder, 2024-06-06:
Thank you for that! Try adding this G-Code to the last line in start G-Code: `G0 F3000 X108.28 Y60.003 Z0.3`

Reevoy24, 2024-06-06:
Sorry I can't find the place where it says start g-code :(

cochcoder, 2024-06-06:
No worries! You can find it in Orca Slicer by clicking on the edit preset next to your printer, go to the `Machine gcode` tab, and Machine start gcode should be right there.

Reevoy24, 2024-06-06:
I tried to print my benchy again in the middle. He started again in the back right corner and then tried to get next to the bed

Reevoy24, 2024-06-07:
@cochcoder i exported my configs from Simplify3D maybe there you find something which could help us.
[Dremel Idea Builder 3D45.txt](https://github.com/user-attachments/files/15744282/Dremel.Idea.Builder.3D45.txt)

cochcoder, 2024-06-07:
@Reevoy24 Thanks for the config, try adding the below values for your bed size: 
[image]

Reevoy24, 2024-06-07:
OMG it works !! so now it would print my benchy in the middle of the plate, but there is no filament flowing out of the nozzle.

cochcoder, 2024-06-07:
> but there is no filament flowing out of the nozzle.

Try disabling `reduce infill retraction`, you'll need to toggle the advance button to see this.

cochcoder, 2024-06-07:
Also please make sure that `use firmware retraction` is disabled (this is found in the printer's settings). Let me know if disabling only one of these things makes it work or if disabling both makes it work.

Reevoy24, 2024-06-08:
I've tried all combinations, both off, just one on or both on, but unfortunately the result is always the same - no fillament comes out

cochcoder, 2024-06-08:
Can you try enabling "use relative e position in the printer's settings?

Reevoy24, 2024-06-08:
I'll try it out on Wednesday

cochcoder, 2024-06-08:
When you try it, and if it works, please try it also with reduce infill retraction

Reevoy24, 2024-06-12:
so the problem was that use relative distances was on. I had to turn it off and delete G92 E0 in the before layer g-code so that it worked. Now it works even if I turn on reduce infill retraction.
The next thing is that the printer does not show the left printing time. 
If we can get this solved, I think we have successfully integrated the printer into the Orca Slicer :)

cochcoder, 2024-06-12:
Does the exported G-Code have the print time estimate in it? If it does, I'm not sure if I can fix the issue with the printing time left. Unless Simplify3D or Digilab exports it in a different format?

Reevoy24, 2024-06-14:
i am not sure if that helps but here you have the G-code from Orca and the G-code from Simplify3D for a TempTower.
Simplify generates to files a g-code file and a g3drem file.
[Orca Slicer SmartTemperatureTower_PLA_180-225_PLA_1h54m -.txt](https://github.com/user-attachments/files/15836908/Orca.Slicer.SmartTemperatureTower_PLA_180-225_PLA_1h54m.-.txt)
[Simplify 3D G-code SmartTemperatureTower_PLA_180-225.txt](https://github.com/user-attachments/files/15836937/Simplify.3D.G-code.SmartTemperatureTower_PLA_180-225.txt)
[Simplify 3D g3drem SmartTemperatureTower_PLA_180-225.txt](https://github.com/user-attachments/files/15836940/Simplify.3D.g3drem.SmartTemperatureTower_PLA_180-225.txt)

cochcoder, 2024-06-14:
@Reevoy24 
Can you please try the newest [build](https://github.com/cochcoder/OrcaSlicer/actions/runs/9517854783) as soon as you can and list any and all issues you have? If you have any issues with temp control, how the printer behaves at the start, or anything to do with the custom G-Code, please include what Simplify3D has. 

I'll continue looking at the files, but does the time left estimate come up when using Simplify3D's gcode, or only with the g3drem file?

Reevoy24, 2024-06-14:
the new Build works perfectly and the time is only shown when i print the g3drem version

cochcoder, 2024-06-14:
Ok, I won't be able to fix the time remaining issue, but at least its working. Thank you all for your help!

Reevoy24, 2024-06-14:
Thank you for your help :)

Reevoy24, 2024-07-24:
@cochcoder hey i found a new problem in the new version i only can use the dremel PLA preset and i cant even add new filaments :( 
[image]

cochcoder, 2024-07-29:
> @cochcoder hey i found a new problem in the new version i only can use the dremel PLA preset and i cant even add new filaments :( ![no_filament](https://private-user-images.githubusercontent.com/102697426/351683332-b6ef116d-1c88-4059-9cf2-cecfc8b795b0.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIyNzMyMTAsIm5iZiI6MTcyMjI3MjkxMCwicGF0aCI6Ii8xMDI2OTc0MjYvMzUxNjgzMzMyLWI2ZWYxMTZkLTFjODgtNDA1OS05Y2YyLWNlY2ZjOGI3OTViMC5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyOVQxNzA4MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNjdhNjcxNzFjNjBjZWZkMWQ0MzJiMTE5ZTZlNWFmMjc5M2Q0NmZlMjdlYzA0OWVjNzNlN2E5Y2I2NDBmM2FkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.XVKc_TS_I4gZSWMVrdDzLqJswJ-s0N-BibCbC25SITg)

Thanks for finding this! I'll look into it as soon as I can!

cochcoder, 2024-09-19:
@Reevoy24 

Sorry for the long wait, I've fixed the issue and started a new build [here](https://github.com/cochcoder/OrcaSlicer/actions/runs/10946923501) if you want to test it out before the next beta release.

cochcoder, 2024-10-26:
@Reevoy24 

Out of curiosity, if you replaced `{input_filename_base}_{filament_type[0]}_{print_time}.gcode` with `{input_filename_base}_{filament_type[0]}_{print_time}.g3drem` in Filename format under the Others tab, does it fix your issues with time remaining and G-Code thumbnails not showing?

Reevoy24, 2024-10-27:
@cochcoder
Unfortunately the Dremel freezes when I insert a stick that contains the g3drem file.

cochcoder, 2024-10-27:
> @cochcoder
> 
> Unfortunately the Dremel freezes when I insert a stick that contains the g3drem file.

That's unfortunate, thanks for trying anyways though

Reevoy24, 2024-10-27:
@cochcoder no problem, but thank you for the idea.

AgentEagle00, 2024-11-07:
> > @cochcoder
> > Unfortunately the Dremel freezes when I insert a stick that contains the g3drem file.
> 
> That's unfortunate, thanks for trying anyways though

yo i’m currently working with a dremel 3d40 and what trying to click your build links but keep getting a 404 error. i’m having pretty much all the same problems as the previous guys but my orca doesn’t have a profile set for dremel on the version im running

cochcoder, 2024-11-07:
> yo i’m currently working with a dremel 3d40 and what trying to click your build links but keep getting a 404 error. i’m having pretty much all the same problems as the previous guys but my orca [...]

The latest version of Orca Slicer has all of the Dremel printers included. I was unfortunately only able to work with others to get the 3D45 working so the 3D40 will probably not work out of the box. If you find any issues and/or solutions to those issues, please feel free to reach out to me and let me know!

AgentEagle00, 2024-11-07:
@cochcoder 

I'm pretty sure the only difference between the 3d40 and the 3d45 is speed and a heated bed. the 3d40 is slower no heated bed and the same build volume. i'm running orca 2.2.0

cochcoder, 2024-11-07:
> @cochcoder 
> 
> I'm pretty sure the only difference between the 3d40 and the 3d45 is speed and a heated bed. the 3d40 is slower no heated bed and the same build volume. i'm running orca 2.2.0 
> 
> and i'm getting the same issue of no extrusion and the print is wayyyy off centered to the top right. 

Would you be able to try some of the fixes posted in this issue to see if they help you as well?

I won't be able to make any fixes on my side for a little while.

AgentEagle00, 2024-11-07:
@cochcoder 

i'll definitely give the the origin change and the "use relative e distance" fix and see what those do for me and let you know.

AgentEagle00, 2024-11-07:
@cochcoder 

so for the 3d40 both of those fixes work just as for the 3d45. only things that are really necessary are the filament presets but i can always make my own filament profiles as i've done for my enders and my positron

cochcoder, 2025-02-28:
@AgentEagle00 

So sorry for not responding sooner. I have applied the origin and "use relative e distance" changes in a separate branch. Can you confirm if the "use relative e distance" has to be disabled after changing the origin?

iantutors-a11y, 2025-11-20:
I am getting an error trying to slice with the Dremel 3D45 config "Error: "G92 E0" was found in before_layer_gcode, which is incompatible with absolute extruder addressing"  Any ideas?

PR #4777 (2024-05-19): Add Dremel 3D20, 3D40, and 3D45
In this PR I added the profiles based as close to the ones in Dremel's slicer, [DigiLab](https://www.dremel.com/ap/en/digilab/software), however, a few QOL settings were added, such as combine infill and Arachne mode. One main concern I have is the start/end G-Codes as I copied them word for word from DigiLab, but I don't know if they will work as intended in Orca Slicer. I think that these printers will make a great addition to Orca Slicer as I only found them in DigiLab and Simplify3D, which DigiLab, in my experience, is based on a very old slicer (if not vers [... cut]