OrcaSlicer/OrcaSlicer #1429: Printer not waiting for first layer bed temperature

silentone, 2023-06-25:
When printing multiple objects sliced in object-by-object mode, and using different bed temperatures, the printer does not wait for the bed to reach the first layer temperature after completing an object and setting the new bed temperature.

Steps:
1. Add 6 objects to build plate
2. Set 1st layer bed temp to 50C & other layers to 30C
3. Configure slicing to go in object by object order.

Expected: After completing each object, issue M140, then M190.

Actual: 
G1 X188.028 Y61.787 Z60.487 F9000 ; move to origin position for next object travel_to_xyz
M140 S50 ; set bed temperature
M104 S205 ; set nozzle temperature
M106 S0 ; disable fan
;LAYER_CHANGE

Windows 10 x64
Orca 1.6.3

cron410, 2023-06-28:
You need to change the M140 and M104 to M190 and M109 respectively in your start gcode. Either within your printer settings in Orca Slicer, or your PRINT_START macro within Klipper configs.

silentone, 2023-06-30:
> > 
> 
> I think the issue opener means the generated gcode of OrcaSlicer. He wants to on object change, moving from layer 79063546325 to layer 1 to heat the bed up again to 50 degrees. and that should be "waited" is his expectations. I doubt his requested solution is manually editing the generated gcode. since it is model dependant, there is nothing you can do by start macro. it is within the print, not at the beginning.

Correct.

closed 2023-11-21 by github-actions[bot]

coryrc, 2025-05-21:
This bug is still present.

SharpEdgeMarshall, 2026-09-30:
I confirm it's still present and it affects not only bed temperature but also nozzle temperature and fan.
The printer doesn't wait and this cause issue with my printing

LJKraus, 2026-10-01:
[Printing by object G-code](https://www.orcaslicer.com/wiki/printer_settings/machine%20gcode/printer_machine_gcode#printing-by-object-g-code)
[¶](https://www.orcaslicer.com/wiki/printer_settings/machine%20gcode/printer_machine_gcode#printing-by-object-g-code)
This G-code is executed at the start of printing each individual object in a multi-object print job. It can be used to set specific parameters or perform actions for each object.

[image]

SharpEdgeMarshall, 2026-10-16:
@LJKraus Yes I know, and that's how I implemented what I consider a workaround.
First layer settings should be automatically applied every time the printer is doing a first layer.

LJKraus, 2026-10-16:
First layer settings apply to Layer One.  

Additional objects begin to print on a different G-Code layer, which is why you have the **_option_** to add by-object code if you wish. 

I've never found that to be necessary, and would not want to wait for temperature changes or risk bed adhesion issues for completed objects.