OrcaSlicer/OrcaSlicer #1435: User Presets Not Saving

Snargola, 2023-06-27:
**Describe the bug*
OrcaSlicer 1.6.3 running on Windows 10
User generated filament profiles are not loading between sessions. I have created a user profile for a particular filament. I save the settings, close the slicer, reopen the slicer and the profile is gone.

**3mf File for This Bug**
There is not .3mf file

**To Reproduce**
Create profile. Save profile. Close slicer. Open slicer

**Expected behavior**
I would expect that the profile would be saved and be available upon restarting the program :)

**Screenshots**

**Printer model**
Bambu X1C

**Desktop (please complete the following information):**
 - OS: Windows 10 Home
 - Version 22H2

six00, 2023-06-28:
这个问题我早就发现了，提交了issue一直没个答复，所有的配置信息 包括机器信息耗材信息 打印设置全都没了

Alexx40in, 2023-06-28:
I have the same problems. More precisely, there are two problems: profiles are reset to default profiles, and those profiles where output scripts are registered are generally deleted

fr3ak2402, 2023-06-28:
Can you send me a file of one of your USER presets?

six00, 2023-06-28:
@Alexx40in @Snargola go back to 1.6.2,everything is fine,the profiles come back!

netweaver1970, 2023-06-28:
I don't dare going to 1.6.3 due to the multiple self resetting/missing/not working profile issues. When I went to 1.6.3 beta, I also lost them, but I somehow managed to get them back by logging in with my Bambulabs account (even if I have a Voron, I still created that account). Maybe lucky.

It might work the same way when going to a final 1.6.3 but for now I'm not sure if I would lose all custom profiles, for sure with all the changes coming up on profile splitting/renaming etc .... So I'm a bit stuck for now till I hear success stories from people upgrading and 'magically' their profiles are still there (or were converted/updated transparently. I rather print and upgrade my printer than fighting with my slicer. Of course, I'm missing all fixes in 1.6.3 too. One day I'll have to bite the bullet...

Snargola, 2023-06-28:
> Can you send me a file of one of your USER presets?
> 
> I would like to take a look at the contents, I already have a guess as to why this is...

I'm not certain what my USER presets are...?

This is the filament profile I have exported. GH won't let me attach the file, so here is a copy/paste:

{
    "additional_cooling_fan_speed": [
        "70"
    ],
    "bed_temperature_difference": [
        "10"
    ],
    "chamber_temperature": "0",
    "close_fan_the_first_x_layers": [
        "1"
    ],
    "compatible_printers": [
        "Bambu Lab X1 Carbon 0.4 nozzle",
        "Bambu Lab X1 0.4 nozzle",
        "Bambu Lab X1 Carbon 0.6 nozzle",
        "Bambu Lab X1 Carbon 0.8 nozzle",
        "Bambu Lab X1 0.6 nozzle",
        "Bambu Lab X1 0.8 nozzle"
    ],
    "compatible_printers_condition": "",
    "compatible_prints": [],
    "compatible_prints_condition": "",
    "cool_plate_temp": [
        "35"
    ],
    "cool_plate_temp_initial_layer": [
        "35"
    ],
    "default_filament_colour": [
        ""
    ],
    "enable_overhang_bridge_fan": [
        "1"
    ],
    "enable_pressure_advance": [
        "1"
    ],
    "eng_plate_temp": [
        "0"
    ],
    "eng_plate_temp_initial_layer": [
        "0"
    ],
    "fan_cooling_layer_time": [
        "100"
    ],
    "fan_max_speed": [
        "100"
    ],
    "fan_min_speed": [
        "100"
    ],
    "filament_cost": [
        "20"
    ],
    "filament_density": [
        "1.24"
    ],
    "filament_deretraction_speed": [
        "nil"
    ],
    "filament_diameter": [
        "1.75"
    ],
    "filament_end_gcode": [
        "; filament end gcode \nM106 P3 S0\n"
    ],
    "filament_flow_ratio": [
        "0.96"
    ],
    "filament_is_support": [
        "0"
    ],
    "filament_max_volumetric_speed": [
        "12"
    ],
    "filament_minimal_purge_on_wipe_tower": [
        "15"
    ],
    "filament_retract_before_wipe": [
        "nil"
    ],
    "filament_retract_restart_extra": [
        "nil"
    ],
    "filament_retract_when_changing_layer": [
        "nil"
    ],
    "filament_retraction_length": [
        "nil"
    ],
    "filament_retraction_minimum_travel": [
        "nil"
    ],
    "filament_retraction_speed": [
        "nil"
    ],
    "filament_settings_id": [
        "kingroon_red_green"
    ],
    "filament_shrink": [
        "100%"
    ],
    "filament_soluble": [
        "0"
    ],
    "filament_start_gcode": [
        "; filament start gcode\n{if  (bed_temperature[current_extruder] >45)||(bed_temperature_initial_layer[current_extruder] >45)}M106 P3 S255\n{elsif(bed_temperature[current_extruder] >35)||(bed_temperature_initial_layer[current_extruder] >35)}M106 P3 S180\n{endif};Prevent PLA from jamming"
    ],
    "filament_type": [
        "PLA"
    ],
    "filament_vendor": [
        "Generic"
    ],
    "filament_wipe": [
        "nil"
    ],
    "filament_wipe_distance": [
        "nil"
    ],
    "filament_z_hop": [
        "nil"
    ],
    "filament_z_hop_types": [
        "nil"
    ],
    "from": "",
    "full_fan_speed_layer": [
        "0"
    ],
    "hot_plate_temp": [
        "55"
    ],
    "hot_plate_temp_initial_layer": [
        "55"
    ],
    "inherits": "Generic PLA",
    "name": "kingroon_red_green",
    "nozzle_temperature": [
        "215"
    ],
    "nozzle_temperature_initial_layer": [
        "210"
    ],
    "nozzle_temperature_range_high": [
        "240"
    ],
    "nozzle_temperature_range_low": [
        "190"
    ],
    "overhang_fan_speed": [
        "100"
    ],
    "overhang_fan_threshold": [
        "50%"
    ],
    "pressure_advance": [
        "0.024"
    ],
    "reduce_fan_stop_start_freq": [
        "1"
    ],
    "required_nozzle_HRC": [
        "3"
    ],
    "slow_down_for_layer_cooling": [
        "1"
    ],
    "slow_down_layer_time": [
        "8"
    ],
    "slow_down_min_speed": [
        "10"
    ],
    "support_material_interface_fan_speed": [
        "-1"
    ],
    "temperature_vitrification": [
        "55"
    ],
    "textured_plate_temp": [
        "55"
    ],
    "textured_plate_temp_initial_layer": [
        "55"
    ],
    "version": "1.6.0.3"
}

fr3ak2402, 2023-06-28:
Okay...

Please open the OrcaSlicer once and close it again after one minute...

Then go to the folder Username/AppData/Roaming/OrcaSlicer/log

Open the log with Notepad++ that matches the time of the last opening and search in this file for the name of your filament profile...
Your profile will probably be displayed as incompatible in the log... or an error message will be written in the log...

Just send the parts of the log where the file occurs...

thisiscam, 2023-06-29:
> Okay...
>
> Please open the OrcaSlicer once and close it again after one minute...
>
> Then go to the folder Username/AppData/Roaming/OrcaSlicer/log
>
> Open the log with Notepad++ that matches the time of [...]

Here's one error I'm getting: 
```can not find parent for config /Users/xxxx/Library/Application Support/OrcaSlicer/user/default/process/0.20mm Standard @Creality TPU.json!
```

thisiscam, 2023-06-29:
> Here's one error I'm getting:
>
> ```
> can not find parent for config
> /Users/xxxx/Library/Application Support/OrcaSlicer/user/default/process/0.20mm Standard @Creality TPU.json!
> ```

The fix is to fix all the `inherits` entries in all your user profiles. 
For example, I needed to change from `inherits: 0.2mm Standard @Creality` to `inherits: 0.2 Standard @Creality Ender3V2`. You might need to look up in the slicer what's the updated  name for your machine.

fr3ak2402, 2023-06-29:
Yes, in your case it is, but in his case unfortunately not, because he has the problem with a user preset based on Bambulab and not on a 3rd party printer...

It would be interesting to know what he says in the user press, which refers to a Bambulab file.

sgttetard, 2023-06-30:
hi, I've noticed that if the "automatically update built-in presets" option is checked, your user presets won't appear. On the other hand, I have to reselect my printer, filament and print settings every time I restart. Is there a way to force my settings to default at startup?

PS: bravo for the work, this slicer is just the best ;)

Alexx40in, 2023-07-02:
confirm. Removed a string containing quotes from post-processing scripts and profiles are no longer deleted and loaded when opening a project

PatMei, 2023-07-04:
> confirm. Removed a string containing quotes from post-processing scripts and profiles are no longer deleted and loaded when opening a project

Yes, that was the trick for me, too. Thank you very much!

foreachthing, 2023-07-05:
Here's another one: https://github.com/SoftFever/OrcaSlicer/issues/1397

Snargola, 2023-07-11:
> Can you send me a file of one of your USER presets?
> 
> I would like to take a look at the contents, I already have a guess as to why this is...

.json file for a machine:
{
    "brim_type": "no_brim",
    "from": "User",
    "inherits": "0.20mm Standard @Creality Ender3V2",
    "initial_layer_infill_speed": "25",
    "initial_layer_speed": "25",
    "inner_wall_speed": "45",
    "is_custom_defined": "0",
    "name": "0.20mm PITA",
    "outer_wall_speed": "35",
    "print_settings_id": "0.20mm PITA",
    "skirt_loops": "0",
    "sparse_infill_density": "5%",
    "travel_speed": "200",
    "version": "1.6.1.0",
    "wall_loops": "2"
}

.json for a filament for the same machine:
{
    "filament_flow_ratio": [
        "1.01772"
    ],
    "filament_max_volumetric_speed": [
        "12.6"
    ],
    "filament_settings_id": [
        "PITA_ziro_gold_silk"
    ],
    "from": "User",
    "hot_plate_temp": [
        "55"
    ],
    "hot_plate_temp_initial_layer": [
        "55"
    ],
    "inherits": "Creality Generic PLA",
    "is_custom_defined": "0",
    "name": "PITA_ziro_gold_silk",
    "nozzle_temperature": [
        "215"
    ],
    "nozzle_temperature_initial_layer": [
        "215"
    ],
    "version": "1.6.1.0"
}

ToledoRayburn, 2023-07-24:
I have tried today creating my own filament profile.    Created and saved twice only to close the slicer and loose everything.   On one hand glad this is a known issue...  What is the status and were are the custom profiles supposed to be saved.

closed 2023-11-21 by github-actions[bot]