OrcaSlicer/OrcaSlicer #1436: [Feature Request] Port Organic Supports from PrusaSlicer

meltiseugen, 2023-06-28:
**Is your feature request related to a problem? Please describe.**
I would like to use the organic supports from PrusaSlicer in OrcaSlicer

**Describe the solution you'd like**
Have a new option for support type tree -> support style -> organic

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Dustmuffins, 2023-07-02:
These are known as Tree supports in Orcaslicer

stonemandave, 2023-07-02:
You will need to tweak the settings quite a bit to imitate Prusaslicer.

YanceyA, 2023-07-03:
> These are known as Tree supports in Orcaslicer

Organic supports are an evolution and different from tree supports in Orcaslicer (which are ported from Cura). Organic should be ported over when the developer has time to review and pull the 2.6 Prusaslicer changes into the Orcaslicer code. 2.6 has heaps of great features to pull over.

Dustmuffins, 2023-07-06:
> Organic supports are an evolution and different from tree supports in Orcaslicer (which are ported from Cura).

After messing around with tree supports more in Orcaslicer, I see what you mean. The organic supports in Prusaslicer are much nicer to use and easier to remove.

YanceyA, 2023-07-30:
I would be willing to offer a modest bounty [~100usd] for this feature port if it could be implemented "soon".

eugenmeltis3pg, 2023-08-12:
> I would be willing to offer a modest bounty [~100usd] for this feature port if it could be implemented "soon".

I would top that with another 50usd.
So now, it would be 150usd for the organic supports.

SoftFever, 2023-08-13:
Hey guys,

Out of curisity, have you guys got a chance to try the slim tree style? 
@eugenmeltis3pg @YanceyA @Dustmuffins 
[image]

Dustmuffins, 2023-08-13:
When I slice with that style, I get unsupported areas with printing in thin air due to not having any interface.

[image]

But I do get properly supported sections using my above settings:

[image]

Dustmuffins, 2023-08-13:
Another bit of info I'd like to add on why I think Organic supports are a worthy inclusion:

I frequently get these strange bugs with the ported Cura Tree supports in Orcaslicer:

[image]

While Prusaslicer doesn't do this:
[image]

One thing I definitely prefer about the tree supports is the ability to use variable layer height, which organic doesn't support in Prusaslicer. 
[image]

If there was a way to modify organic to be able to do this, that would be the perfect support for 90% of my cases IMO. It may be prohibitively difficult though, or else I'd imagine prusaslicer would already do it.

However, as great as Prusaslicer is, Orcaslicer is just so incredible so I've been putting up with this slight annoyance 😁

closed 2023-11-28 by Noisyfox