OrcaSlicer/OrcaSlicer #1437: [Feature Request] [MacOS] Is it possible to port the Repair Model feature from PrusaSlicer?

meltiseugen, 2023-06-28:
**Is your feature request related to a problem? Please describe.**
Context: MacOS
Some models need repairs, PrusaSlicer does this automatically. In OrcaSlicer I get a notification saying that the repair feature is only available for the Windows distribution. 

Can it also be available for MacOS distribution? (Since PrusaSlicer on MacOS has it)

**Describe the solution you'd like**
To be able to repair models in OrcaSlicer on a MacOS distribution

**Describe alternatives you've considered**
Repairing in PrusaSlicer, then exporting the model, and then importing it in OrcaSlicer

**Additional context**
N/A

jphoke, 2023-06-29:
Second this -- it is beyond inefficient if I need to go fire up a Win VM move a model file, fix it and move it back ... this can (and should) be ported over to one of my Macs I use. Should be in Orca (hell it should be in BS too but I'd take orca over BS tbh)

MikeOxlong83, 2023-07-03:
I also vote for this, nothing but laziness is preventing us from also being able to use the repair feature!

DOWN WITH MEANINGLESS OPERATING SYSTEM DISCRIMINATION!!!!!!

jphoke, 2023-07-03:
@MikeOxlong83  but seriously. It's not laziness on either side of this -- I just hope full a macOS version that has full support of the slicer functionality ... and my pronouns are Mac / Nix :)

Highlander-bdj, 2023-09-16:
Bump, I too would LOVE Z this!

closed 2023-12-24 by github-actions[bot]

ajaytanna, 2024-04-14:
I too would like this sorely missed feature. Firing up a windows machine for just this - ouch!