OrcaSlicer/OrcaSlicer #1459: !! Move exceeds maximum extrusion (93.311mm^2 vs 0.640mm^2)

gaifeng8864, 2023-07-04:
**Describe the bug**

I used the following project and slicing presets to slice and send to the printer to start printing, but the printer gave me an error message when printing to the second layer:!! Move exceeds maximum extrusion (93.311mm^2 vs 0.640mm^ 2). Obviously, this is surprisingly wrong.
After viewing the Gcode file. I found that somewhere in the second layer, there are obvious abnormal action instructions in the Gcode file, just like the one in the picture.

[image]

**3mf File for This Bug**

[Exported preset files.zip](https://github.com/SoftFever/OrcaSlicer/files/11944783/Exported.preset.files.zip)

[handle.zip](https://github.com/SoftFever/OrcaSlicer/files/11944964/handle.zip)

**To Reproduce**

Use the preset file in "Exported.preset.files.zip" to slice "handle.3mf" in "handle.zip", and you will find abnormal Gcode commands in the second layer.

**Expected behavior**

Fix this bug. In addition, it is better to add a Gcode code pre-running function, or a Gcode inspection function to facilitate finding and locating the cause of the error.

**Printer model**

DIY machines, coreXY models. The "max_extrude_cross_section" option is the default.

**Desktop (please complete the following information):**
 - OS:
Version Windows 10 Pro
Version number 22H2

 - Version [e.g. 22]
 
 OrcaSlicer 1.6.3
[image]

StarkRG, 2023-07-09:
I'm having the same issue. There's also something screwy going on with the volumetric flow rate scale in the preview mode. I'm pretty sure that's a different color there, but with a scale like that it's impossible to say for sure.
[image]
Weirdly, I can't find anything that isn't dark blue so I'm not sure how it's determining the scale. 
[image]

Deselecting "Detect narrow internal solid infill" gives a slightly different result, but still unprintable.
[image]

I thought it might be related to wiping so I disabled that function which got yet another different, unprintable result.
[image]

Switching to classic wall generatior, more different, still unprintable.
[image]

Narrow infill back on. There just seems to be no pattern I can detect.
[image]

It's not speed related.
[image]

Or width.
[image]

Sineos, 2023-07-09:
Seen the same. Seems only happening when using absolute extrusion. In relative extrusion mode, the same file worked for me.

gaifeng8864, 2023-07-09:
@StarkRG 
@Sineos 
It does appear that this is a problem. I then sliced several times, but the printing stopped after an error was reported halfway. I have temporarily switched to Superslicer and it works without any issues. However, it is known that it has not been updated for a long time, which is very regrettable. Superslicer is a very good slicing software, it has brought me a lot of help, more than 90% of my slicing projects are done with it, I really hope this project can continue.

madsi1m, 2023-07-17:
I have the same problem, fixed it by setting fan kick starts to 0 and problem went away

gaifeng8864, 2023-07-17:
> I have the same problem, fixed it by setting fan kick starts to 0 and problem went away

Surprising solution. Do you mean disable the fan all the way?

madsi1m, 2023-07-17:
No no  still have the fan on, but I had to set these two values to 0s
[image]

gaifeng8864, 2023-07-18:
> No no still have the fan on, but I had to set these two values to 0s [image]

Oh, this value seems to be the time to control the opening advance of the fan. Actually this value will introduce this bug, I really didn't expect it. Thank you for the information you provided, I will also set this value to 0 to try.

gaifeng8864, 2023-07-21:
> No no still have the fan on, but I had to set these two values to 0s [image]

After my test, it is indeed as you said. After turning off these two options the flow error disappeared. I read the descriptions of these two options, currently they only support G0 and G1 commands, initially I suspect that they conflict with the G2 arc command, but when I turn off the arc fitting, the problem of abnormal flow will still appear.

madsi1m, 2023-07-21:
it might be better to set kick-start time in klipper rather than slicer. (speed-up should be in slicer if needed.)

gaifeng8864, 2023-07-21:
> it might be better to set kick-start time in klipper rather than slicer. (speed-up should be in slicer if needed.)

What exactly does "kick-start time" mean?

zurajm, 2023-07-23:
> After my test, it is indeed as you said. After turning off these two options the flow error disappeared. I read the descriptions of these two options, currently they only support G0 and G1 commands, [...]

Thanks guys, setting these to 0 works for me also. Definitely, this is a bug.

madsi1m, 2023-07-23:
> > it might be better to set kick-start time in klipper rather than slicer. (speed-up should be in slicer if needed.)
> 
> What exactly does "kick-start time" mean?

Runs fan at full speed for set time then drops to correct fan speed. Some bigger fans or lots of fans on the same output might struggle to output (for example) 20% speed and fan will just not move. But if you spin at 100% for say 1s then drop to 20% the fan will spin at 20% because it is already moving.

Something you should set in klipper under [fan] rather than in slicer...

From klipper config page:
```
#kick_start_time: 0.100
#   Time (in seconds) to run the fan at full speed when either first
#   enabling or increasing it by more than 50% (helps get the fan
#   spinning). The default is 0.100 seconds.
```

I'm using x2 5015 blowers and have set kick_start_time: 0.400 (in klipper) and 0s in slicer

gaifeng8864, 2023-07-24:
> Runs fan at full speed for set time then drops to correct fan speed. Some bigger fans or lots of fans on the same output might struggle to output (for example) 20% speed and fan will just not move. [...]

Thank you for your detailed answer, it made me finally understand what this concept means. I'm going to set it up like this to see the effect.

gaifeng8864, 2023-09-06:
As of the version "OrcaSlicer V1.6.6 Official Release", this problem has not been fixed yet.

cynixx3, 2023-10-30:
I also had the Move exceeds maximum extrusion error during a print with a klipper printer that was solved by setting the cooling fan speed-up time and kick-start time to 0. Using Orca Slicer 1.8.0-beta1 on linux with a Klipper SV07 unmodded printer. Thank you to @madsi1m for finding the temp fix.

gaifeng8864, 2023-10-31:
> I also had the Move exceeds maximum extrusion error during a print with a klipper printer that was solved by setting the cooling fan speed-up time and kick-start time to 0. Using Orca Slicer 1.8.0-beta1 on linux with a Klipper SV07 unmodded printer. Thank you to @madsi1m for finding the temp fix.

Yes, this is a good temporary solution for now and I've been using it. However, I really hope the developers can fix this bug.

hatkyinc2, 2023-11-18:
sounds the same as the issue https://github.com/SoftFever/OrcaSlicer/issues/2033

StarkRG, 2023-11-29:
Unfortunately, I just ran into this bug again even with kick-start and speed-up times set to zero.

closed 2024-04-17 by github-actions[bot]