OrcaSlicer/OrcaSlicer #1462: Add a translation into Russian.

stasyan68, 2023-07-04:
Translation files in the attachment. Thanks
[RU.zip](https://github.com/SoftFever/OrcaSlicer/files/11950030/RU.zip)

iStarComLab, 2023-07-04:
> Translation files in the attachment. Thanks
> [RU.zip](https://github.com/SoftFever/OrcaSlicer/files/11950030/RU.zip)

Thank you for your hard work! Checked on 1.6.3 everything is ok!

stasyan68, 2023-07-04:
> > Файлы перевода во вложении. Спасибо
> > [RU.zip](https://github.com/SoftFever/OrcaSlicer/files/11950030/RU.zip)
> 
> Спасибо вам за вашу тяжелую работу! Проверено на 1.6.3, все в порядке!

I failed to build binaries for Windows. Can you help?

iStarComLab, 2023-07-04:
> > > Файлы перевода во вложении. Спасибо
> > > [RU.zip](https://github.com/SoftFever/OrcaSlicer/files/11950030/RU.zip)
> > 
> > 
> > Спасибо вам за вашу тяжелую работу! Проверено на 1.6.3, все в порядке!
> 
> I failed to build binaries for Windows. Can you help?

Sorry, but I only use Macos!

Alexx40in, 2023-07-05:
> Translation files in the attachment. Thanks [RU.zip](https://github.com/SoftFever/OrcaSlicer/files/11950030/RU.zip)

Создал папку RU, положил туда файлы - в списке языков русский не появился
___
I created a folder RU, put files there - Russian did not appear in the list of languages

stasyan68, 2023-07-05:
> > Файлы перевода во вложении. Спасибо [RU.zip](https://github.com/SoftFever/OrcaSlicer/files/11950030/RU.zip)
> 
> Создал папку RU, положил туда файлы - в списке языков русский не появился
> 
> Я создал папку RU, поместил туда файлы - русский не появился в списке языков

Это так не работает, нужно скомпилировать .EXE, а перед этим добавить данные файлы в исходный код. у меня не получается(

iStarComLab, 2023-07-24:
та же тема https://github.com/SoftFever/OrcaSlicer/issues/757

iStarComLab, 2023-08-06:
РЕШЕНО!

closed 2023-11-21 by github-actions[bot]