OrcaSlicer/OrcaSlicer #1463: Custom Bed Textures don't work

ThatCamOp, 2023-07-04:
**Description**
When adding a texture the build plate turns black instead of taking on the texture color.

**To Reproduce**
Click the edit printer button, under printable area hit set. Then upload the STL and either of the textures provided below and save.

[Bed.zip](https://github.com/SoftFever/OrcaSlicer/files/11950686/Bed.zip)

**Expected behavior**
The bed should be customized in the shape of the stl and the texture provided

[image]

**Ender 3 Pro**

**Windows 10 Home**

ItsDeidara, 2023-08-03:
Same issue here. I've tried both SVG and PNG files with no luck

The  "model" section seems to work and render the STL properly however. I am using an AMD Radeon RX 6700S

kluupom, 2023-09-01:
I do have the same issue with several printers that came with the slicer.
E.g. Voxelab Aquila, all Artillery printers.
Some, like the prusa MK3 work perfectly fine.
I tested it with three different computers: only one can show the buildplate.

the biggest difference between the computers is the make of the graphics card:
Nvidia seems to work, AMD doesn’t…

ItsDeidara, 2023-11-28:
The fix for AMD users is to resize the image to a power of 2. I used this nifty tool. There might be better ways however as this made my image size HUGE. But hey whatever works lmao

https://github.com/RyanAWalters/PowerOf2ImageResizer

closed 2024-03-06 by github-actions[bot]