OrcaSlicer/OrcaSlicer #1467: Can't import configs I exported

sprior, 2023-07-05:
**Describe the bug**
I wanted to create a customized Voron profile that I don't have to rename from the standard one.  What I've done with PrusaSlicer is to load the standard profile, save it as a user profile, export it, then remove that printer using the printer config and import it from my exported config.

I tried this procedure with Orca and was able to export the printer/filament/print configs.  I then removed the inherits lines.  When I try to import those configs it just says 0 configs imported without telling me why.  I've also changed the config names slightly (renamed Voron to Voron1 everywhere in the file and the filenames) and that didn't make any difference - nothing imported.

**To Reproduce**
Steps described above

**Expected behavior**
The exported configs should be imported and they're not.

**Printer model**
Started with the 0.4 Voron 350 config and tried to export/modify/import it

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 1.6.3
[configs.zip](https://github.com/SoftFever/OrcaSlicer/files/11953146/configs.zip)

HamXSlamwich, 2023-07-07:
I am also having this issue, needed to reinstall windows on my PC and when I tried to load in configs that I had just exported, Orca Slicer says "There are 0 configs imported. (Only non-system and compatible configs)"

thebigone2087, 2023-07-10:
Also having this error. Was wanting to import my PrusaSlicer config that I have tuned well for my Voron, but it will not import the .ini file that is created. Hoping for a fix.

anoccomir, 2023-09-25:
Have this issue too, trying to import settings for Tenlog printers using different bed leveling settings and material profiles, but we cant seem to be able to import them with this error coming up every time, I used to be running the latest non beta build of the software but i hoped switching to 1.7.0 beta would give this problem a solution, well it didnt, and I cant get settings i need on my pc. Help?

Peck07, 2023-10-26:
Check the compatible_printers, inherits attibute values in the profiles.
I've added my printer to the comatible_printers list, inherits also includes the printer name or vendor, when I've changed these to be "compatbile" with my current settings, the import succeeded. (before the changes, I've got the 0 configs imported too)

sprior, 2024-01-25:
This issue is still current as far as I am concerned.

bluestrike2, 2024-02-01:
Bumping this after running into the same issue.

closed 2024-05-09 by github-actions[bot]