OrcaSlicer/OrcaSlicer #1473: wall_infill_order behaving incorrectly, flips wall order

unixcarbide, 2023-07-06:
**wall_infill_order behaving incorrectly, flips wall order**
When using `wall_infill_order` in the `outer/inner/infill` or `inner/outer/inner/infill` settings, the settings are reversed at various laayers, nearly randomly.

The result can produce stripes of a sort of *blob-banding*, whereby the layer inner/outer order is reversed for one layer.
These *blob-bands* do not continue around the entire print, they just taper out.

https://github.com/SoftFever/OrcaSlicer/assets/130264439/738d9546-95ff-485e-92af-f8f65488f1b1
[image]
[image]
[image]
[image]

**3mf File for This Bug**
With a seam on the "back", it is easy to flip through the `wall_infill_order` settings, and to see all the various unretract and seam positions..

**To Reproduce**
Steps to reproduce the behavior
Load a file like the tower with seam I have attached.
Set the seam to "Back", (or any big flat vertical wall).
Try different settings of `wall_infill_order`, what OrcaSlicer shows in the sliced view, will print with *blob bands*.

**Expected behavior**
The `wall_infill_order` setting `outer/inner/infill` behaves exactly as expected, (in contrast to `inner/outer/infill` which does not behave as expected).

Here is an image just after I figured this out:
[image]

**Screenshots**
[image]
[image]

**Printer model**
FLSun SR, stock-ish V6 hotend with volcano nozzle.  Extruder is a bowden setup, with OMG v2 extruder.
Running Klipper.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version 13.2.1
 - OrcaSlicer 1.6.3

unixcarbide, 2023-07-06:
One anecdotal note, (I have not replicated for this test):

After looking back at some old prints, when I was a SuperSlicer user, I found exactly the same artifacts in box like objects.  I believe this could be an issue going back to some common ancestor in the software.

The760, 2023-07-07:
I am having a very similar issue in #1468 

I went back into the slicer to check for anything weird going on around certain layers, and it seems that the issue is indeed that the wall order is getting flipped around from what I have it set on.

Noisyfox, 2023-07-23:
I think this is an issue of the Arachne wall generator. I could not reproduce this issue using Classic.

closed 2023-11-21 by github-actions[bot]