OrcaSlicer/OrcaSlicer #1494: Rotating the view should rotate around the viewpoint rather than the centre of the bed

StarkRG, 2023-07-08:
**Is your feature request related to a problem? Please describe.**
If you zoom in and move the viewpoint camera to look at a problematic part of your print, and then rotate to look at it from another angle, you will no longer be looking at the part of the print you were previously looking at and will have to move the camera again.

**Describe the solution you'd like**
At the _very_ least, there should be an option to choose whether to rotate around the bed or the camera (or, rather, a point somewhere in front of the camera), but, really, rotating around the camera should be the default.

**Describe alternatives you've considered**
I can't imagine an alternate solution.

**Additional context**
Something weird is going on here, so zoom in for a btter look.
[image]
And rotate a teensy bit... that's not the same spot at all.
[image]

StarkRG, 2023-07-08:
Here's how rotating currently works in OrcaSlicer:
https://github.com/SoftFever/OrcaSlicer/assets/451657/8dece1bb-55f6-4068-b349-0c580ca09f0f

Here's how it works in SuperSlicer:
https://github.com/SoftFever/OrcaSlicer/assets/451657/e1d4aada-cbf7-4ae4-bdf1-017fd2263d9d

KD7LRJ, 2023-07-08:
Hold Ctrl to lock the rotation point.
Hold Alt to allow rotation while pointing at an object.
Hold Ctrl+Alt to lock the rotation point and rotate around it while pointing at an object.

I wish I could set Ctrl+Alt as the default and have the key combos for working the other way.

StarkRG, 2023-07-08:
Ahh, thanks for that. Gotta love hidden features. /s

Still, as you say that should be the default behavior.

StarkRG, 2023-11-06:
It's not so much "stale" as it is "being ignored by the devs"

KD7LRJ, 2024-02-12:
I'm still interested in having an preference setting for this. Just the three of us so far though. Anyone else?

nixmeer, 2024-03-17:
Me. It is so annoying to change the view angle in orca slicer. I cannot imagine anybody using the rotation feature and thinking "I like that"...

closed 2024-06-24 by github-actions[bot]