OrcaSlicer/OrcaSlicer #1495: [Klipper] N.O. Layer not showing in Klipper

blacktoz, 2023-07-09:
**Describe the bug**
I am using Mainsail, with Moonraker interface on a rp2040, and no informations can be seen with Orcaslicer, such as slicer time, and the most important for me, the Layer number, it shows layer 0 of 0.

**3mf File for This Bug**
Welp, this bug happens to every .stl / .3mf that I've used

**To Reproduce**
Slice the file, upload it to klipper, launch the print, and layer number won't show and other informations about the slicing.

**Expected behavior**
Sliced files should show these informations

**Screenshots**
[image]
[image]

**Printer model**
Creality Ender 3 

**Desktop (please complete the following information):**
 - Arch Linux
 - 6.4.2
 - OrcaSlicer 1.6.3

Sineos, 2023-07-10:
See https://docs.mainsail.xyz/overview/slicer/prusaslicer#layer-count

SoftFever, 2023-07-10:
Orca Slicer is not recognized by Moonraker judging from your screenshot.
Most likely your Moonraker is not up to date.
Can you update it and check again?

`3. Klipper users, please update Moonraker to the latest version so that it can recognize OrcaSlicer and parse metadata like thumbnails.`

blacktoz, 2023-07-10:
> See https://docs.mainsail.xyz/overview/slicer/prusaslicer#layer-count

Thanks, that worked perfectly

> Orca Slicer is not recognized by Moonraker judging from your screenshot. Most likely your Moonraker is not up to date. Can you update it and check again?
> 
> `3. Klipper users, please update Moonraker to the latest version so that it can recognize OrcaSlicer and parse metadata like thumbnails.`

Yeah, it has been a while that I haven't updated these packages, i'll do that, thanks!

closed 2023-07-10 by blacktoz