OrcaSlicer/OrcaSlicer #1496: How to distribute custom vendor profiles

nordurljosahvida, 2023-07-09:
Prusaslicer supports distributing custom profiles via the vendor folder, which allows a user to then add an unofficial printer profile via the setup wizard.

It doesn't yet support then updating them automatically [see https://github.com/prusa3d/PrusaSlicer/issues/6780], but at least it's possible.

Does orca support this? I tried installing the vendor directory inside the data dir but it doesn't reflect that in the wizard.

I know bambu supports syncing via cloud, but i've never tried that.

How can we distribute custom profiles - for our organization's custom printer farm - inside orca?

Thanks!

SoftFever, 2023-07-10:
Have you tried export config?
That should work

nordurljosahvida, 2023-08-26:
Hi, thanks, the idea here is to be able to actually push out to our organization a vendor folder with all of our profiles inside. Export config then needs to be manually imported every time and is not the same thing, as with vendor the printer profiles are treated as master and official.

To make everything simpler, I was hoping orca would sync our custom printers:

[image]

but others in the team using orca logged in to bambu account don't see any custom machines.

Is this a limitation of the sync backend or of orca?

Thanks!