OrcaSlicer/OrcaSlicer #1500: "Keep fan always on" will prevent the aux fan from starting.

XSvFury, 2023-07-09:
Hi, 

It seems the "keep fan always on" option will prevent the "m106 p2 s255" command from being included in the g-code file. If unselected, the command will present and the aux fan will work as intended. 

Just in case there is some other conflict I am unaware of, I have included screenshots of my printer and filament settings.
[image]
[image]

Kylemraley, 2023-07-10:
Curious if this affects chamber fan as well. I have my p3 gcode set to run chamber fan at full speed. I have this “keep fan on” setting checked, but regardless of my gcode edits it will only run at 40%. When I manually change in my device tab, it sets itself to 10%. Will only go to 40% and no higher when I turn off chamber fan, then back on.

shyblower, 2023-08-22:
This bug is still present in 1.6.4-beta3

KD7LRJ, 2023-10-23:
Confirmed this is still an issue with OS 1.7.0.

[image]

[image]

[Hexagon Diffusion_pr03 - Copy.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13079773/Hexagon.Diffusion_pr03.-.Copy.3mf.zip)

closed 2024-01-30 by github-actions[bot]