OrcaSlicer/OrcaSlicer #1505: Jerk Control Not Working (RepRapFirmware)

CCS86, 2023-07-10:
With Jerk Control enabled using RRF output, only 2 jerk commands appear in the whole gcode output (M205)

[image]

[image]

[Orca_Jerk.3mf.txt](https://github.com/SoftFever/OrcaSlicer/files/11996604/Orca_Jerk.3mf.txt)

CCS86, 2023-10-03:
?

BR7408, 2023-11-06:
I found the exact same issue. M205 command generated is X7 Y7, no matter what jerk settings I put in the console, unless I set default jerk to 0.

Is this going to get fixed any time soon? It would be a huge benefit to have variable jerk capability on RepRap.

igorscheffer, 2023-11-24:
I had a similar problem, I discovered it was the printer settings. for example, the maximum jerk was set to 9, so when I put something higher in the filament configuration it wouldn't

closed 2024-03-01 by github-actions[bot]